Oros42/IMSI-catcher #1: there is only wrong imsi numbers

aleen42, 2016-05-05:
- When I use the condition: `p[71:][:2]=='\x08\x29'` and `p[62:][:2]=='\x08\x29'`, I have get no any imsi numbers output, and I think I have set the correct frequency.
- When I use a condition: `p[71:][:2]=='\x08\x49'` and `p[62:][:2]=='\x08\x49'`, there are some numbers output on the screen, but they all are not belong to any devices I want to track. So what can I do with this catcher?

Oros42, 2016-05-05:
Sorry, I'm not en expert in GSM and I have test it only on a french network.  
But first step to investigate is to use wireshark :  
```
sudo wireshark -k -Y '!icmp && gsmtap' -i lo
```
And if you can send me a wireshark capture, it would be useful to debug.

aleen42, 2016-05-06:
So, May I ask you a question, and can I also get IMEI from the gsm package?

Oros42, 2016-05-06:
Ooh IMEI and ISMI are not the same thing.
ISMI is used to identify the user (SIM card). It's not link to the phone https://en.wikipedia.org/wiki/International_mobile_subscriber_identity
IMEI is a number to identify the phone. And it's not link to the SIM https://en.wikipedia.org/wiki/IMEI

In my wireshark sniffing, I've never seen IMEI.

aleen42, 2016-05-06:
According to some suggestions, IMEI may not be transmitted with GSM packages and of course we won't get this number in the sniffed package, unless the situation that service providers has asked phones for IMEI. :cry:

Oros42, 2016-05-06:
Yes.
You can found your own ISMI number to track your phone but it's not easy.
This is how I process : 
1- check that your phone use GSM (2G) network
2- run gqrx (http://gqrx.dk/)
3- with your phone phone, call a second phone and find which frequency is used to upload (phone to antenna, around 876Mhz to 914Mhz)
4- if you get the frequency, add an offset of 45Mhz (ex : 880Mhz->924Mhz)
5- use this frequency with my program and «normally», you will have your ISMI in the list of ISMI found.
6- repeat step 1 to 5 in other network area and check if you have see a common number in the list

good luck

aleen42, 2016-05-06:
Ok, I'll check your process and see whether it's okay, Thx.

stenroot, 2016-09-14:
Hello, today published an updated program in GNU Radio release v3.7.10.1, how to you can add your IMSI-Catcher?

Oros42, 2016-09-14:
Sorry, I don't understand this part "how to you can add your IMSI-Catcher" :-/

SenseProg, 2016-09-14:
Hi, 

 

We need UHF reader with radius-polarisations, up 30 db/m transmit power, up 6-10 m (minimum 5) recommend read range, frequency range -- 860Mhz-875Mhz . With four ports for ANT.

And also we need UHF reader with around (radius) polarisations antenna  up 9 dbi. 

Their size must be up 250x250. 

Must be readed 200-300 tags in one moment. 

 

Can you propose something? 

 

Thank you very much for answers!

 

Best regards, 

 

Bogdan Parfenyuk, 

Embedded developer,

SenseSystems

Oros42, 2016-09-14:
Wooow WTF?
I have just made this script for fun with the USB DVB-T key RTL2832U. I didn't sell anything.

aleen42, 2016-09-14:
:-), WTF, an ad email?

stenroot, 2016-09-22:
Hello I'm using GNU Radio v3.7.10.1(http://gnuradio.org/news/gnu-radio-v3-7-10-1-release/) installed on the USB, how to add your IMSI-Catcher?

Oros42, 2016-09-23:
Mmmh I haven't try to setup GNU Radio without using pybombs :-/

Oros42, 2016-09-26:
@stenroot : I have found a new way to setup gr-gsm :
```
sudo add-apt-repository -y ppa:ptrkrysik/gr-gsm
sudo apt update
sudo apt install gr-gsm python-numpy python-scipy python-scapy
```

Oros42, 2016-09-29:
I have made some fixes in the code.

SonyKucing, 2016-09-29:
i am in indonesia with MCC : 510 and MNC : 10
where code is to edit for correct imsi catcher ?

Oros42, 2016-09-29:
MCC and MNC are store in mcc-mnc/mcc_codes.json
mcc-mnc/update_codes.py is used to update mcc-mnc/mcc_codes.json with data from https://en.wikipedia.org/wiki/Mobile_Network_Code

SonyKucing, 2016-09-29:
work perfectly bro... thanks for update your code...

onoff0, 2016-10-15:
perfect, I can find many IMSI, but also forcing my cell phone in 2G mode I can not find my IMSI. I also tried to analyze with wireshark on the downlink frequency of my phone but nothing. you have ideas?

aleen42, 2016-10-15:
@onoff0 This catcher will always lose some IMSIs, when it's not included in GSM headers. According to some experts, IMSIs won't be transferred between phones and base stations after the connection at the first time. Therefore, I advise you to try to force to break the connection between them, so that phones can transfer IMSIs again to reconnect.

Oros42, 2016-10-15:
@onoff0 You should check if 2G and 3G use the same frequency.

onoff0, 2016-10-17:
thanks for the answers. I make some tests I noticed that the phone in GSM mode when LAC changes or switching on can wait for an update. Just then an SMS to force the network to request a location update, and then take its IMSI.

onoff0, 2016-10-17:
it would be possible to implement the catcher for TMSI, with a grouping counter of unique values? thanks !!

Oros42, 2016-10-17:
Work in progress for TMSI since last week ;-)
Coming soon.

onoff0, 2016-10-17:
great !!! good job ;)

aleen42, 2016-10-18:
@onoff0 During forcing network to request a location update, IMSIs will be transferred in the GSM package?

onoff0, 2016-10-18:
@aleen42 Yes, 
personally I have only tried it with phones in GSM. with wireshark you can see in detail the network packets received from mobile phones under a specific frequency assigned by the telephone operators. The IMSIs newly assigned (change LAC, telephone turned on, location update) will be visible in wireshark via a paging request, after which the network will assign to mobile phones in the same area a TMSI that will remain so for some time.

Oros42, 2016-10-22:
Hello everyone !
I have add the display of TMSI ! :-)
And bonus, you can filter IMSIs and follow one special IMSI by using this :
```
sudo python IMSI-TMSI-catcher.py -m "123 45 6789101112"
```
or
```
sudo python IMSI-TMSI-catcher.py -m 123456789101112
```

onoff0, 2016-10-23:
great, I'm trying.
I do not mean a lot of programming, it would be possible to implement a function that picks up only the TMSI in a given frequency with the possibility to filter only those that repeat more. example, only displays the TMSI that is repeated in 'area for more than 3 or 4 times ++.
Congratulations for your work!

stevevaius, 2016-10-31:
Hi, for an NGO project, we are looking a way to determine the number of mobile cells around a radius. We need to count the unique mobiles via detecting their IMSI or something unique. Your tool looks like built for it. How we can implement? Any help appreciated!

Oros42, 2016-11-01:
@onoff0 : If you displays the TMSI that is repeated X times, you can't have a live TMSI tracking. You should wait X time to sniff GSM network and after that, count each TMSI.
I have thinking about filtering one TMSI. But It's not easy. The TMSI could change during the time.

@stevevaius : The variable nb_IMSI already count the number of unique IMSIs (column 1). But I don't known yet how to count mobiles that haven't send IMSI and use only TMSI.

rdarioc, 2016-11-24:
Guys, I am impressed by the quality of your work. @Oros42 , is it possible, based on your experience, to access the SS7 network of the mobile operator? My main concern is privacy (geolocation and packet sniffing and decoding) and I am wondering if we need to worry ONLY from the improper use of the SS7 protocols accessed by an hub provided by the mobile operator or if we should ALSO worry from an "anonymous" access done with this type of technology. Thank you. p.s. I understand this may not be the right place for this conversation and I apologize for it, so please, move this message where it more appropriately fits.

Oros42, 2016-11-24:
Thanks you @rdarioc :-)
This issue is more like a topic than a true issue XD
I'm sorry, I don't know yet who does SS7 works but I want to learn it !
I need documentation about SS7.

rdarioc, 2016-11-24:
Ask, and you shall receive: 

Signalling System #7, ITU-T Q.700 series:
http://www.itu.int/rec/T-REC-Q/e

Mobile Application Part (MAP) specification, 3GPP TS 29.002:
http://www.3gpp.org/ftp/Specs/archive/29_series/29.002/

Study into routeing of MT-SMs via the HPLMN, 3GPP TR 23.840:
http://www.3gpp.org/ftp/Specs/archive/23_series/23.840/

Support of Optimal Routeing (SOR), 3GPP TS 23.079:
http://www.3gpp.org/ftp/Specs/archive/23_series/23.079/

[25c3-locating-mobile-phones.pdf](https://github.com/Oros42/IMSI-catcher/files/612111/25c3-locating-mobile-phones.pdf)

Oros42, 2016-11-25:
Ooh thanks :-D

habibur333, 2017-01-02:
Is it possible to retrieve phone numbers or MSIN  from IMSI numbers?

Oros42, 2017-01-02:
You can't have the phone number but you have MSIN in the IMSI number.
MSIN is the last part of the IMSI.
https://en.wikipedia.org/wiki/International_mobile_subscriber_identity

habibur333, 2017-01-03:
Thanks Oros42 for your fast response. here is 2 issue
1st issue:
I am trying to collect phone number from air. is there any way to collect them? 
2nd issue
from this link https://en.wikipedia.org/wiki/International_mobile_subscriber_identity  For Bangladesh the  MSIN  number and phone number is same not sure for other countries. at present there is no such type of imsi number in Bangladesh for grammin phone which exampled in wiki.

Oros42, 2017-01-03:
If it's true that MSIN number and phone number is same in Bangladesh then my IMSI-catcher got it !
In my country (France), the phone number is not in the MSIN.

BlackPhreaker, 2017-02-02:
Good day ... In my opinion it would be nice to add the output log !!!

Oros42, 2017-02-02:
Hello
You can do this :
```sh
sudo -s
python IMSI-TMSI-catcher.py > log.csv&
tail -f log.csv
```
;-)

BlackPhreaker, 2017-02-02:
It is possible and so I agree !!! But I was referring to the script, with the date and time ...

Oros42, 2017-02-02:
This ?
```
date > log.csv
python IMSI-TMSI-catcher.py >> log.csv&
tail -f log.csv
```
:-D

BlackPhreaker, 2017-02-02:
Perhaps esle Could you be able to help me here in this example ...

***
def write_log(x):
	file_object = open("./gsm.log", "a+")
	try:
		file_object.write(x)
	finally:
		file_object.close()
***

Not like it is impossible to add to your script ((

Oros42, 2017-02-02:
Add your function in my code and change «print» to «write_log» at lines 172, 174 and 185.

BlackPhreaker, 2017-02-02:
I can not do ... But still does not work ... Check please !!!

***
#!/usr/bin/python
# -*- coding: utf-8 -*-

import time
from scapy.all import sniff
import json
from optparse import OptionParser

imsis=[] # [IMSI,...]
tmsis={} # {TMSI:IMSI,...}
nb_IMSI=0 # count the number of IMSI

# Log Start
def write_log(x):
file_object = open("./gsm.log", "a+")
try:
file_object.write(x)
finally:
file_object.close()
# Log End

# return something like '0xd9605460'
def str_tmsi(tmsi):
	if tmsi != "":
		new_tmsi="0x"
		for a in tmsi:
			c=hex(ord(a))
			if len(c)==4:
				new_tmsi+=str(c[2])+str(c[3])
			else:
				new_tmsi+="0"+str(c[2])
		return new_tmsi
	else:
		return ""

# return something like '208 20 1752XXXXXX ; France ; Bouygues ; Bouygues Telecom'
def str_imsi(imsi, p=""):
	new_imsi=''
	for a in imsi:
		c=hex(ord(a))
		if len(c)==4:
			new_imsi+=str(c[3])+str(c[2])
		else:
			new_imsi+=str(c[2])+"0"
	
	mcc=new_imsi[1:4]
	mnc=new_imsi[4:6]
	country=""
	brand=""
	operator=""
	if mcc in mcc_codes:
		if mnc in mcc_codes[mcc]['MNC']:
			country=mcc_codes[mcc]['c'][0]
			brand=mcc_codes[mcc]['MNC'][mnc][0]
			operator=mcc_codes[mcc]['MNC'][mnc][1]
			new_imsi=mcc+" "+mnc+" "+new_imsi[6:]
		elif mnc+new_imsi[6:7] in mcc_codes[mcc]['MNC']:
			mnc+=new_imsi[6:7]
			country=mcc_codes[mcc]['c'][0]
			brand=mcc_codes[mcc]['MNC'][mnc][0]
			operator=mcc_codes[mcc]['MNC'][mnc][1]
			new_imsi=mcc+" "+mnc+" "+new_imsi[7:]
		else:
			country=mcc_codes[mcc]['c'][0]
			brand="Unknown"
			operator=mcc_codes[mcc]['MNC'][mnc][1]
			new_imsi=mcc+" "+mnc+" "+new_imsi[6:]
[... 184 lines cut ...]

parser = OptionParser(usage="%prog: [options]")
parser.add_option("-a", "--alltmsi", action="store_true", dest="show_all_tmsi", help="Show TMSI who haven't got IMSI (default  : false)")
parser.add_option("-i", "--iface", dest="iface", default="lo", help="Interface (default : lo)")
parser.add_option("-m", "--imsi", dest="imsi", default="", type="string", help='IMSI to track (default : None, Example: 123456789101112 or "123 45 6789101112")')
parser.add_option("-p", "--port", dest="port", default="4729", type="int", help="Port (default : 4729)")
(options, args) = parser.parse_args()

show_all_tmsi=options.show_all_tmsi
imsi_to_track=""
if options.imsi:
	imsi="9"+options.imsi.replace(" ", "")
	for i in range(0,15,2):
		imsi_to_track+=chr(int(imsi[i+1])*16+int(imsi[i]))

# mcc codes form https://en.wikipedia.org/wiki/Mobile_Network_Code
with open('mcc-mnc/mcc_codes.json', 'r') as file:
	mcc_codes = json.load(file)

print("{:7s} ; {:10s} ; {:10s} ; {:17s} ; {} ; {} ; {}".format("Nb IMSI", "TMSI-1", "TMSI-2", "IMSI", "country", "brand", "operator"))
sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi, store=0)

write_log(str(time.time()) + '\n')

***

BlackPhreaker, 2017-02-02:
Can add your script log many think it will come in handy in the future ... For Rania sposibo for pomasch !!!

Oros42, 2017-02-02:
You need to indent your code.
This :
```python
# Log Start
def write_log(x):
file_object = open("./gsm.log", "a+")
try:
file_object.write(x)
finally:
file_object.close()
# Log End
```
should be :
```python
# Log Start
def write_log(x):
    file_object = open("./gsm.log", "a+")
    try:
    file_object.write(x)
    finally:
    file_object.close()
# Log End
```

Sorry, but I'm not sure to understand your last message :-s

BlackPhreaker, 2017-02-02:
I mean ... Thank you for helping me ...

Oros42, 2017-02-02:
Okay ;-)

BlackPhreaker, 2017-02-02:
Currently tuning a script ... If you want I can send you then that ye have put him at ???

BlackPhreaker, 2017-02-22:
Hi Oros42 !!! Thank you for your previous help ... There was such Problem I can not add on your display example (LAC) and (CID) ... Esle Could you not mogliby help ???

Oros42, 2017-02-22:
Oooh yes, I've got a script for that. I clean it and publish it soon ;-)

BlackPhreaker, 2017-02-22:
It has helped me very much ... Thank you'll be waiting !!!

Oros42, 2017-02-22:
I've add find_cell_id.py and immediate_assignment_catcher.py.
Could you try it?

BlackPhreaker, 2017-02-23:
Thank you very much ... your code works perfectly !!! I hope I have for your example will create a script with the final conclusion ...

===================================================================
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI     ; Country    ; Brand ; Operator  ; MCC   ; MNC ;  LAC   ; CID    ;
===================================================================

BlackPhreaker, 2017-02-23:
And also add the output ARFCN MHz dBm and all this in one script!!!

BlackPhreaker, 2017-02-23:
Oooh There was a lot of problems combining pre script ... It's a pity I will do next

BlackPhreaker, 2017-02-23:
Tell me what I am doing wrong ???

https://github.com/BlackPhreaker/GSM/blob/master/IMSI_Catcher_Ver1.0.7.py

BlackPhreaker, 2017-02-23:
The head has burst !!!

[6 more replies]

closed 2017-07-26 by Oros42