Oros42/IMSI-catcher #3: IMSI-catcher for OsmocomBB ?

osmo-guy, 2016-10-01:
Hello :)
I am happy about your tool and it is working perfect, even if I do not have gr-gsm, but just regular airprobe.
My request would be to write just the same program for osmocombb.
If you have a Motorola phone you can test it. 
My idea would be to run osmocoms command  "./ccch_scan -a 871 -i 127.0.0.1"
and your tool will collect/extract all the IMSIs from ARFCN 871, in this example.
Are you up for it?

Best regards,
Alex

Oros42, 2016-10-01:
Sorry, I haven't got a Motorola and I haven't test OsmocomBB :-s
I have you try run my scrip and "./ccch_scan -a 871 -i 127.0.0.1" at the same time ?

I have you try to run "./ccch_scan -a 871 -i 127.0.0.1" and open wireshark ?
And have you found "Message Type: Paging Request Type 1" or "Message Type: Paging Request Type 2" with IMSI number ?

osmo-guy, 2016-10-04:
>  I have you try run my scrip and "./ccch_scan -a 871 -i 127.0.0.1" at the same time ?
Yes and your script cant adapt to it :-/

> I have you try to run "./ccch_scan -a 871 -i 127.0.0.1" and open wireshark ?
And have you found "Message Type: Paging Request Type 1" or "Message Type: Paging Request Type 2" with IMSI number ?

Yes, Wireshark shows me all of the Message Types.

Is there a possibility to extract the clear IMSIs from Wireshark and display them so nicley in the terminal like in your previous script ?

Oros42, 2016-10-04:
Okay
With Wireshark, could you paste here a copy of a packet how have an IMSI ?
It's something like this :
```
0000   00 00 00 00 00 00 00 00 00 00 00 00 08 00 45 00
0010   00 43 1c d4 40 00 40 11 1f d4 7f 00 00 01 7f 00
0020   00 01 c2 e4 12 79 00 2f fe 42 02 04 01 00 00 00
0030   c9 00 00 16 21 26 02 00 07 00 31 06 21 00 08 XX
0040   XX XX XX XX XX XX XX 2b 2b 2b 2b 2b 2b 2b 2b 2b
0050   2b
```
XX XX XX XX XX XX XX XX = IMSI
(Other possibilities are in the code)

You could change the 3 last bytes.

bastienbaranoff, 2016-10-04:
0000   00 00 00 00 00 00 00 00 00 00 00 00 08 00 45 00
0010   00 43 1c d4 40 00 40 11 1f d4 7f 00 00 01 7f 00
0020   00 01 cb bc 12 79 00 2f fe 42 02 04 01 00 00 00
0030   d1 00 00 25 3d ea 05 00 00 00 31 06 21 00 08 29
0040   26 10 56 04 40 3X XX 2b 2b 2b 2b 2b 2b 2b 2b 2b
0050   2b

XXX = last 3 bits of the IMSI

osmo-guy, 2016-10-04:
Sorry, posted with a wrong account, content is correct though :)

Oros42, 2016-10-04:
29 26 10 it's 262 01 ( Germany ; Telekom ; Telekom Deutschland GmbH)
https://en.wikipedia.org/wiki/Mobile_Network_Code
I don't understand why this not work :-/

osmo-guy, 2016-10-04:
I think your script handles osmocom output different, than that from airprobe/sdr.
Maybe we can teamview and you can check it yourself? Do you have a private email ?

bastienbaranoff, 2016-10-04:
Do you see avalaible ok at the end of the ouput of osmonitb?

Oros42, 2016-10-04:
I haven't setup teamview client but you can contact me by email or XMPP at oros[at]ecirtam.net

closed 2016-10-05 by Oros42