Oros42/IMSI-catcher #4: Open issue for any questions

Oros42, 2017-02-02:
Post here your questions about my IMSI-catcher.

oscarmh, 2017-05-16:
Hi there regarding the Key issue https://github.com/Oros42/IMSI-catcher/issues/8 it is all about a lost MCC/MNC ? or any other issue ? Best regards

oscarmh, 2017-06-16:
... What about LTE ? Any suggestions ? KR.

Oros42, 2017-06-17:
You should find a LTE receiver because gr-gsm could only receive GSM.
I haven't search yet.

scaery, 2017-07-26:
Hi, thanks for the goods... ehm, why i am unable to track my own IMSI? It never shows up it the logs? I have 4 cells around me for my provider. I switched my phone to 2G only and disabled data. The phone jumps sometimes from one to another cell but still my IMSI won`t show up...

Tool from Play Store to get my IMSI: Network Info II 

I can see many IMSI`s there but filtering out my own with "-m" switch leaves me empty. Im switching the cellid`s manually with  airprobe_rtlsdr.py when i see the phone somehow changed the frequency and jumps to another cell, so why there is nothing in the logs? Do you know a better way to track your own?

Oros42, 2017-07-26:
I have no idea yet :-/

raghurammailbox, 2017-09-02:
getting this error 
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC   ; MNC  ; LAC   ; CellId
WARNING: Unless called manually, this could indicate deprecated use. Should be changed to bytes(self)
WARNING: Unless called manually, this could indicate deprecated use. Should be changed to bytes(self)

raghurammailbox, 2017-09-02:
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC   ; MNC  ; LAC   ; CellId
WARNING: Unless called manually, this could indicate deprecated use. Should be changed to bytes(self)
WARNING: Unless called manually, this could indicate deprecated use. Should be changed to bytes(self)

Oros42, 2017-09-02:
Which version of python did you use ?

raghurammailbox, 2017-09-03:
Python 3.5.2

Oros42, 2017-09-03:
We have plan to rewrite the code. So in waiting of this you can run it with python 2.7.

alexandr84, 2017-09-05:
Hello, this software looks promising. However I can't get it to work with grgsm_livemon. If I start simple_IMSI-catcher.py, grgsm_livemon throws this error: 

Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 270, in <module>
    tb = grgsm_livemon(fc=options.fc, gain=options.gain, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff, args=options.args)
  File "/usr/local/bin/grgsm_livemon", line 157, in __init__
    self.blocks_socket_pdu_0_0 = blocks.socket_pdu("UDP_SERVER", "127.0.0.1", "4729", 10000)
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig5.py", line 1062, in make
    return _blocks_swig5.socket_pdu_make(type, addr, port, MTU, tcp_no_delay)
RuntimeError: bind: Address already in use

If I start grgsm_livemon it runs fine but then simple_IMSI-catcher.py results in this error:

Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 535, in <module>
    udpserver(port=options.port, prn=find_imsi)
  File "simple_IMSI-catcher.py", line 488, in udpserver
    sock.bind(server_address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use

It looks like I can't get to start both things at the same time. Any tips? 

Thank you!

Oros42, 2017-09-05:
Try this :
```
sudo python simple_IMSI-catcher.py -s
```

alexandr84, 2017-09-05:
It worked, thank you!

Wallace78, 2017-09-21:
Hi, thank you for a great script!

Iḿ running Ubuntu 16.04 LTS on a virtualbox. Nothing else except update and uprade has been done. I have installed gr gsm and IMSI catcher according to the following instructions;

`sudo apt-get install git python-pip`

`sudo pip install PyBOMBS`

`sudo pybombs prefix init /usr/local -a default_prx`

`sudo pybombs config default_prefix default_prx`

`sudo pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git`

`sudo pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
`
`sudo pybombs install gr-gsm`

`sudo ldconfig`

`sudo apt install python-numpy python-scipy python-scapy`
`git clone https://github.com/Oros42/IMSI-catcher.git `
`python IMSI-catcher/mcc-mnc/update_codes.py`

I can start the IMSI-catcher script with `sudo python simple_IMSI-catcher.py` (also tried with `simple_IMSI-catcher.py -s`)
However when trying to run grgsm_livemon from terminal 2 i get the following error:

Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000,052982 Hz
[R82XX] PLL not locked!
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 370, in <module>
    main()
  File "/usr/local/bin/grgsm_livemon", line 358, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/local/bin/grgsm_livemon", line 191, in __init__
    self.blocks_socket_pdu_0_0 = blocks.socket_pdu("UDP_SERVER", "127.0.0.1", serverport, 10000, False)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig5.py", line 419, in make
    return _blocks_swig5.socket_pdu_make(*args, **kwargs)
RuntimeError: bind: Address already in use 
`
Any help regarding this matter is much appreciated! 
/Wallace

Oros42, 2017-09-24:
```python simple_IMSI-catcher.py``` works only with the last gr-gsm (https://tracker.debian.org/pkg/gr-gsm).
But ```sudo python simple_IMSI-catcher.py -s``` should works 0_o

Could you try this :
- update IMSI-catcher
- in term 1 run ```sudo python simple_IMSI-catcher.py -s```
- in term 2 run ```grgsm_livemon```

Sollai, 2017-09-27:
Hi Have problem using "python simple_IMSI-catcher.py -m". It returns mistake

Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 559, in <module>
    if imsi_to_track_len%2 == 0 and imsi_to_track_len > 0 and imsi_to_track_len <17:
NameError: name 'imsi_to_track_len' is not defined

Tried to write -m 123456789101112 and "123 45 6789101112" (example) , always this mistake
How can I fix it?

Oros42, 2017-09-27:
It should be fixed.

Sollai, 2017-09-27:
Yep, fine now, thanks!

Wallace78, 2017-09-28:
@Oros42 
Cheers and sory for not replying sooner!
 I reinstalled everything and got it to work with the command  `python simple_IMSI-catcher.py -s` . However I am experiencing some issues.

1. I´m a little suspicious that all the captures are not correct. I capture more foreign IMSI  (Austria, Germany and Guam (!?) that should be present in my area a few of them might be right but looking at other users captures, this seems to be an issue for others as well. 

2. Also, I´m not able to capture my own IMSI (The phones are set on 2g and I switch between flightmode and active) I try this to confirm wether I am actually capturing correct and existing IMSI. As mentioned, no luck.

Any suggestions regarding this?

The third issue is most likely due to limited knowledge when it comes to linux and is not most likely related to IMSI-catcher.. After a run with grgsm_livemon I´m not able to restart the grgsm. I recieve the error `RuntimeError: bind: Address already in use`. I have to reboot the system to run grgsm again.

For others that experience the same problems with 0 (overflows?) Lowering the sample rate to 1M fixed my issue and I was able to capture IMSI. Inititally  ( `grgsm_livemon -s 1M `) 

Once again, thank you for this!

/Wallace

Oros42, 2017-09-28:
1- It's possible that there are errors. To check that, you can run ```wireshark``` and looking for ```Message Type: Paging Request Type 1``` packets.

2- I quite sure that flightmode doesn't completly turn off baseband. So you should remove the barrery.

3- I think you haven't correcly exit ```grgsm_livemon```. If the command ```pgrep grgsm_livemon``` return you a number, then you have ```grgsm_livemon``` running in backgroud.
So you should use ```kill -15 <Number_returned_by_pgrep>```.

DRLXX, 2017-10-01:
Hi, I am very interested in your work,  is this program can demodulate the GSM1800 band Or get  information of GSM1800? Thank you very much!

Oros42, 2017-10-02:
It's not my program who made the demodulation. It's grgsm who made it.
And yes, it can demodulate the GSM1800.

DRLXX, 2017-10-03:
@Oros42 thank you very much , Your program IMSI-catcher  gave me a lot of help for my study, sincerely thank you!

Micolocobr2, 2018-03-05:
Hi,
Im installed everything without errors but when I try to run I got the error:

marcelo@imsi:~/Downloads$ sudo python simple_IMSI-catcher.py --sniff
  File "simple_IMSI-catcher.py", line 7
    <!DOCTYPE html>
    ^
SyntaxError: invalid syntax
marcelo@imsi:~/Downloads$ 

I got the same error on version of gr-gsm >= 0.41.2-1

Could you help me please?

FFY00, 2018-03-06:
@Micolocobr2 post the output of

```
sudo python -V
```

Micolocobr2, 2018-03-06:
FFY00
Thanks for your response but I realize that when I download the files from Github, then came with some html code inside. I edited it and solved the problem.

tks

FFY00, 2018-03-06:
@Micolocobr2 you probably were using the wrong python version. The reason it worked it's not because you edited the html files but because for some reason the program was called with a different python version.

Mortensv, 2018-03-06:
HI

When i run "sudo idconfig" it gives me the error: "sudo: idconfig: command not found"

any ideas as to why this happens

Oros42, 2018-03-06:
```idconfig``` 0_o ???
Where did you see this ?
Perhaps it's ```ifconfig``` you search ?

Mortensv, 2018-03-07:
[image]
i don't understand why this happens

Mortensv, 2018-03-07:
could it be because of some errors i got when i intalled it

FFY00, 2018-03-07:
@Micolocobr2 what are you trying to do?

Micolocobr2, 2018-03-07:
Hi,

I dont know why but all the files I download was with peaces of html code inside.
Once I remove it within the files, problem solved!

Thanks guys for your help.

FFY00, 2018-03-07:
@Micolocobr2 how are you downloading the files??

Micolocobr2, 2018-03-08:
From de github in the oros42 page using Firefox from linux 16.04 desktop.
If you click on each file to download, it comes with something wrong in the code. Some html piece of code.

At the time I was not realize the "clone or download" button, then I tried to download files one by one.

FFY00, 2018-03-09:
@Micolocobr2 please download the files from this link https://github.com/Oros42/IMSI-catcher/archive/master.zip

SigPloiter, 2018-05-24:
Hello @Oros42 appreciate your work :), wanted to know, you sniff and decode paging requests ? because there are times that TMSI and IMSI appear together so i assumed that they are paging requests, so just wanted to make sure on which type of traffic you extract the IMSIs

thanks

Oros42, 2018-05-24:
In simple_IMSI-catcher.py, from line 426 to 522 you have comments who explain which type of packet I use.
Packets was extracted with wireshark.
;-)

thewizkid87, 2018-05-31:
Hi @Oros42 

Thanks for sharing this, i have learned a ton about GSM from your code.

My question is where did you find the data on the structure of the packets. I want to understand more in depth the whole flow of data.

I read about frames and all that, but how did you find that:
ord(p[0x12]) == 0x1b:  # Message Type: System Information Type 3
for example
or any of the other parsing of the packet.

If you send me a link to the recourse or whatever would be very much appreciated.

Thanks

Oros42, 2018-05-31:
Oooh fuck. The documentation didn't follow the upgrade of the code :-(
There are an offset of 0x2a.
0x12 (from the code) + 0x2a (offset) == 0x3c (in documentation's dump)

thewizkid87, 2018-05-31:
Just pulled and saw the changes.

Ahhh ok, the offset makes a lot more sense... 

Im still wondering where you found the structure from? how did you know what address is what data?

Thanks

Oros42, 2018-05-31:
I only use wireshark to understand how packets are made.

godlelo, 2018-12-04:
Hi Oros42, first of all many thanks for sharing this development I will try to run it. I have just one doubt, this solution is compatible with macOS ?

Cheers

Oros42, 2018-12-04:
I don't know. I only use Gnu/Linux.
You have to check if you can setup gr-gsm on macOS.
https://osmocom.org/projects/gr-gsm/wiki/Installation

godlelo, 2018-12-04:
Many thanks for your quick feedback mate, I will check then

ezevu, 2019-01-09:
Hi Oros42, I already installed everything (I think) but when I tried to check my antenna with rtl_test it gives me the following information:

Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
No supported tuner found
Enabled direct sampling mode, input 1
Supported gain values (1): 0.0 
Sampling at 2048000 S/s.
No E4000 tuner found, aborting.
Reattached kernel driver

when I try to run the grgsm_scanner -b GSM900 -g 40 -d
I got the following error:

Args= 
gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.14.0.0-220-g97935b15
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
No supported tuner found
Enabled direct sampling mode, input 1
Exact sample rate is: 2000000.052982 Hz
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 426, in <module>
    main()
  File "/usr/local/bin/grgsm_scanner", line 423, in main
    options.ppm, options.gain, options.args, prn = printfunc, debug = options.debug)
  File "/usr/local/bin/grgsm_scanner", line 328, in do_scan
    ppm=ppm, gain=gain, args=args)
  File "/usr/local/bin/grgsm_scanner", line 237, in __init__
    self.wideband_receiver = wideband_receiver(OSR=4, fc=carrier_frequency, samp_rate=sample_rate)
  File "/usr/local/bin/grgsm_scanner", line 144, in __init__
    self.__init(OSR, fc, samp_rate)
  File "/usr/local/bin/grgsm_scanner", line 163, in __init
    100)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/filter/pfb.py", line 71, in __init__
    self._oversample_rate)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/filter/filter_swig.py", line 4384, in make
    return _filter_swig.pfb_channelizer_ccf_make(numchans, taps, oversample_rate)
RuntimeError: Failed to create FFTW wisdom lockfile: /home/tekil/.gr_fftw_wisdom.lock
Reattached kernel driver
corrupted size vs. prev_size
Aborted (core dumped)

I am doing something wrong? Is the frequency wrong? 
Can you support?

PS: I tested my antenna in windows 10 with blazehd and it is working.

Many thanks

phamduythai92mta, 2019-01-21:
is it possible to know how many phones connecting to a basestation?

Oros42, 2019-01-21:
@ezevu I don't know. Ask to @ptrkrysik (https://github.com/ptrkrysik/gr-gsm)

@phamduythai92mta you could have an idea of how many by counting IMSI from the output of my program. But you can't have the exact number.

phamduythai92mta, 2019-01-22:
thanks for the answer, but i am still getting confused that the IMSI is rarely transmitted to BS, so how can i get my phone's IMSI?

bmp51, 2019-04-28:
kinda new to SDR's here. Picked up a NooElec R820T SDR &  DVB-T NESDR mini.  I followed the guide got stuff installed (running on Raspbian (PI)). when I run ` python simple_IMSI-catcher.py` no errors, just looks like its waiting to display data.  when I start `python scan-and-livemon` I get 

> *** Error in `python': corrupted double-linked list: 0x0193e250 ***
Aborted

> 
So not sure if the antenna is not being detected or if I borked up the install? Any ideas?

ptrkrysik, 2019-05-02:
Hi all,

There is no such thing as passive IMSI catcher. You can get some of the
IMSIs transmitted by the network, but only on some relatively rare
occasions, for some small percentage of current users.

The whole purpose of IMSI catcher is to get all IMSIs, so one can do
nefarious stuff like i.e. pinpointing IMSI to a given person handset
based on capturing all IMSIs in places where a given person is expected
to be, or getting IMSIs of all people attending a protest.

Assuring that you get most of IMSIs in the area (and not some small
fraction of them) can be assured only by performing active attack.

bmp51, 2019-05-05:
re-installed on ubuntu no errors I just get nothing back... I can query the antenna no problem (followed manf guidelines to load correct drivers etc).  so I guess progress?  Going to try a different antenna see if I get anything new....  Question will the scan pyhton script run up and down the frequency list or will it simply pick a default MGHTZ?

zoxb, 2019-05-27:
How can resolve this problem with imsi catcher  
 “Python can not open file ‘simple imsi -catcher .py’
Why this error appear ?

dydfrancis, 2020-03-09:
please can i intercept a specific phone number with this imsi tool fydfrancis10@gmail.com

sdr-skip, 2020-05-07:
Hi there,

Im trying to get the IMSI catcher working for several times now and it failed every time. . So i decided to install it again and do it step by step according this site https://osmocom.org/projects/gr-gsm/wiki/Installation

Now i come to the point that i have to download and install gr-gsm and do the cmake .. command. and that is where the shit started to hit the fan.....

And i'm getting this information from CMakeError.log

Can someone explain me what i have to do to fix this?

Thanx!

Determining if the pthread_create exist failed with the following output:
Change Dir: /home/sdr/Desktop/sdr/gr-gsm/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_f066e/fast"
/usr/bin/make -f CMakeFiles/cmTC_f066e.dir/build.make CMakeFiles/cmTC_f066e.dir/build
make[1]: Entering directory '/home/sdr/Desktop/sdr/gr-gsm/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_f066e.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_f066e.dir/CheckSymbolExists.c.o   -c /home/sdr/Desktop/sdr/gr-gsm/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_f066e
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_f066e.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_f066e.dir/CheckSymbolExists.c.o  -o cmTC_f066e 
CMakeFiles/cmTC_f066e.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_f066e.dir/build.make:97: recipe for target 'cmTC_f066e' failed
make[1]: *** [cmTC_f066e] Error 1
make[1]: Leaving directory '/home/sdr/Desktop/sdr/gr-gsm/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_f066e/fast' failed
make: *** [cmTC_f066e/fast] Error 2

File /home/sdr/Desktop/sdr/gr-gsm/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/sdr/Desktop/sdr/gr-gsm/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_e014d/fast"
/usr/bin/make -f CMakeFiles/cmTC_e014d.dir/build.make CMakeFiles/cmTC_e014d.dir/build
make[1]: Entering directory '/home/sdr/Desktop/sdr/gr-gsm/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_e014d.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_e014d.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.10/Modules/CheckFunctionExists.c
Linking C executable cmTC_e014d
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_e014d.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_e014d.dir/CheckFunctionExists.c.o  -o cmTC_e014d -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_e014d.dir/build.make:97: recipe for target 'cmTC_e014d' failed
make[1]: *** [cmTC_e014d] Error 1
make[1]: Leaving directory '/home/sdr/Desktop/sdr/gr-gsm/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_e014d/fast' failed
make: *** [cmTC_e014d/fast] Error 2

Oros42, 2020-05-07:
I'm not the author of gr-gsm !
Have you try this https://osmocom.org/projects/gr-gsm/wiki/Installation#Installation-from-packages-on-Debian-Testing-and-Ubuntu-1804 ?

sdr-skip, 2020-05-08:
> I'm not the author of gr-gsm !
> Have you try this https://osmocom.org/projects/gr-gsm/wiki/Installation#Installation-from-packages-on-Debian-Testing-and-Ubuntu-1804 ?

So you mean i have to start from that point instead of the beginning of the page?

Oros42, 2020-05-08:
I recommend you to READ ALL the page. And you will notice there are different way to install depending your setup.

GianfrancoRocco, 2020-06-08:
Hey Oros, could you give me a hand? I'm trying to execute both simple_IMSI-catcher.py and grgsm_livemon, but I can't execute one if the other is running because they're using the same address, or port, I don't know. The error message is the in both terminals:
`RuntimeError: bind: Address already in use`
I don't know what am I supposed to do with that, soulnd't I be supposed to be able to run both without a problem? Is there any way to solve this?
I would really appreciate your help with this.

Oros42, 2020-06-08:
Did you do :
```
sudo python3 simple_IMSI-catcher.py --sniff
```
?
See https://github.com/Oros42/IMSI-catcher#with-an-old-version-of-gr-gsm

[20 more replies]