Oros42/IMSI-catcher #6: Understanding how it works

ranshalit, 2017-04-10:
Hello,

I made some reading, but I still not sure about some fundamentals:
I would please like to ask, please:
Does rtl-sdr device also transmit (for this catcher to work), or is it used only as a receiving ?
Can I use any rtl-sdr (I am not sure all capable of transmit) ?

Thank you very much!

Oros42, 2017-04-11:
Hello,
My IMSI-Catcher is a passive system (==only receiving).
So you can use any rtl-sdr device.

ranshalit, 2017-04-22:
Can it be used together with openBTS ?

I viewed https://www.youtube.com/watch?v=fQSu9cBaojc demonstration, but couldn't understand how it can catch outbound calls , because as far as I understand , openBTS can't connect with commercial network(?), only with other openBTS networks. 

Thx

Oros42, 2017-04-22:
I have never used openBTS. So it not easy for me to answer you.
If openBTS can send data to 127.0.0.1 then yes you could use my IMSI-catcher.

For «openBTS can't connect with commercial network(?)», I don't know.

closed 2017-07-26 by Oros42

Keskebeu, 2018-02-09:
Yes you can connect I have (to test it). And have used all the current (active) cellid, with with a free android app, and this app sniffing imsis and also gain control of the radio frequency so the phone actually restarts on the same frequency. I was also able to send sms to imsi not timsi on the network because capturing the original imsi on active mode. I used the opposite of the frequency from 1900 to 850 and still sniffed it actively. So be careful don't mess up... Its real. 

Once the commercial tower has known your credentials it tells your phone to not connect to your tower because its not approved by their policy and doesnt allow authentication with your tower anymore. 

You can try on test mode. But also that its becoming limited.