Oros42/IMSI-catcher #7: NOT AN ISSUE: SUGGESTION

1kali2kali, 2017-04-17:
Sir will it be nice to suggest me how to combine the info en IMSI with the find_cell so I can display the MCC/MNC/LCC info with the IMSI located, can you please ? Thanks in advance

Oros42, 2017-04-17:
Hello
The easy way :
All informations of find_cell_id.py are link to the frequency (and your location). Same things for simple_IMSI-catcher.py.
So if you run simple_IMSI-catcher.py and find_cell_id.py in parallel at one frequency then all IMSI you got have the same cell tower (same MCC/MNC/LCC).

Complicate way :
Put the code of find_cell from find_cell_id.py in find_imsi from simple_IMSI-catcher.py after ligne 188.
And fix print().

1kali2kali, 2017-04-17:
sweeeet I will got thru complicate way.... sounds better !! Thanks !!

Oros42, 2017-04-17:
Okay :-)
Please let me know how you will display MCC/MNC/LCC.
I am interested ;-)

oscarmh, 2017-04-17:
wow its more tricky than I suppose......

1kali2kali, 2017-04-17:
sure..... what´s you approach ?

1kali2kali, 2017-04-17:
[image]
N00b!!!! More tomorrow !!!

1kali2kali, 2017-04-18:
Blocked.... any suggestion for show_imsi in order to include MCC/MNC/LAC ??

Oros42, 2017-04-18:
Cell informations are not send every time so you can store MCC/MNC/LAC in a global variable.
But if you change the frequency, MCC/MNC/LAC will not be immediately update :-/

1kali2kali, 2017-04-18:
[image]
;-) working !! and yessss changes LAC/CellID when change freq

Oros42, 2017-04-18:
Good job :-)
Could you share your code ?

1kali2kali, 2017-04-18:
oh my gosh master ask for it........ attached !! 
[imsipluscellfinder.py.txt](https://github.com/Oros42/IMSI-catcher/files/930103/imsipluscellfinder.py.txt)

Oros42, 2017-04-18:
Thanks.
I will update my code soon :-)

Oros42, 2017-04-18:
Merge done.
But what I say previously, it's not perfect when you change the frequency.
[image]

1kali2kali, 2017-04-18:
d#mn something we can do on that ?.....

Oros42, 2017-04-18:
Perhaps catching the switch of frequency from airprobe_rtlsdr.py ?
Or patch airprobe_rtlsdr.py for sending the frequency on network ?

closed 2017-07-26 by Oros42

E3V3A, 2017-07-26:
@Oros42 

> Perhaps catching the switch of frequency from airprobe_rtlsdr.py ?
> Or patch airprobe_rtlsdr.py for sending the frequency on network ?

Why did you close this? Did you actually implement it?

Oros42, 2017-07-26:
It's not related to the initial question and the issue is old without activity.
So this points need a new issue. But I have no idea of who to resolve it.