Oros42/IMSI-catcher #8: Error

ricanara, 2017-04-24:
Hola amigos buenos dias, al tratar de correr la orden sudo python simple_IMSI-catcher.py
me saca el siguiente error:

WARNING: No route found for IPv6 destination :: (no default route?)
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country ; brand ; operator ; MCC   ; MNC  ; LAC   ; CellId
Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 416, in <module>
    sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi, store=0)
  File "/usr/lib/python2.7/dist-packages/scapy/sendrecv.py", line 586, in sniff
    r = prn(p)
  File "simple_IMSI-catcher.py", line 286, in find_imsi
    find_cell(x)
  File "simple_IMSI-catcher.py", line 273, in find_cell
    operator=mcc_codes[mcc]['MNC'][mnc][1]
KeyError: '10'

Por favor, alguien me puede ayudar?
Gracias.

Oros42, 2017-04-24:
I fixed the bug.
Please report me the unknown MNC.
Perhaps I need to update mcc-mnc/mcc_codes.json

ricanara, 2017-04-25:
Esta perfecto.  Muchas gracias!!

Oros42, 2017-04-25:
All IMSI and TMSI you can see are phones who are connected to the antenna.

ricanara, 2017-04-27:
Hola nuevamente, realice algunas capturas muy bien, pero luego me empezo a salir el siguiente error:

WARNING: No route found for IPv6 destination :: (no default route?)
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country ; brand ; operator ; MCC   ; MNC  ; LAC   ; CellId
Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 420, in <module>
    sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi, store=0)
  File "/usr/lib/python2.7/dist-packages/scapy/sendrecv.py", line 586, in sniff
    r = prn(p)
  File "simple_IMSI-catcher.py", line 341, in find_imsi
    show_imsi(imsi1, imsi2, tmsi1, tmsi2, p)
  File "simple_IMSI-catcher.py", line 187, in show_imsi
    print("{:7s} ; {:10s} ; {:10s} ; {} ; {:4s} ; {:5s} ; {:6s} ; {:6s}".format(str(n), str_tmsi(tmsi1), str_tmsi(tmsi2), str_imsi(imsi1, p), mcc, mnc, lac, cell))
ValueError: Unknown format code 's' for object of type 'int'

Me puedes ayudar?

Gracias

Oros42, 2017-04-27:
Oups
Bug fixed.

ricanara, 2017-04-27:
It's perfect thank you very much. Any concerns will be published.
Greetings from Colombia

Oros42, 2017-04-28:
;-)

ricanara, 2017-05-13:
Hola amigo, recientemente empece a tener este error con el dongle rtl-sdr

grgsm_livemon -f890500000 -g 40
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-162-g2790b51f
gr-osmosdr v0.1.4-91-gcf954948 (0.1.5git) gnuradio 3.7.12git-94-gb44e38ce
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000,052982 Hz
[R82XX] PLL not locked!
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
rtlsdr_read_async returned with -5
Violación de segmento (`core' generado)

No he podido escanear las frecuencias por este error, me puedes ayudar a corregir?

Oros42, 2017-05-15:
Check your USB connection with the rtl-sdr dongle.
This error could happen when you unplug the rtl-sdr dongle.

oscarmh, 2017-05-15:
y tambien chequea que tengas un hub de USB con una fuente externa

ricanara, 2017-05-21:
Todo esta bien.

Me pueden orientar que puedo hacer con los IMSI una vez capturados?, es decir esta informacion para que nos puede servir?

Gracias

Keskebeu, 2018-03-08:
It usually looks for python-scapy when that happens in most cases is a unstable install of gui. A reboot did it for me.

closed 2019-01-20 by Oros42