Oros42/IMSI-catcher #9: find_cell_id_.py is not working

aguilasainz, 2017-06-01:
Hi there @Oros42,

Your utility is so great! I've tried to run the extra tool 'find_cell_id.py', but I'm getting this error:

```
ubuntu@ubuntu:~/IMSI-catcher$ sudo python find_cell_id.py 
WARNING: No route found for IPv6 destination :: (no default route?)
MCC   ; MNC  ; LAC   ; CellId ; Country ; Brand ; Operator
Traceback (most recent call last):
  File "find_cell_id.py", line 88, in <module>
    sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_cell, store=0)
  File "/usr/lib/python2.7/dist-packages/scapy/sendrecv.py", line 586, in sniff
    r = prn(p)
  File "find_cell_id.py", line 75, in find_cell
    operator=mcc_codes[mcc]['MNC'][mnc][1]
KeyError: '05'

```

Any idea? Thanks!

Oros42, 2017-06-02:
Hello
No time to fix this today but you can take a look to https://github.com/Oros42/IMSI-catcher/commit/85067041da9804cc0ae8ab328b1861872ece3bd0
It's the same bug.

aguilasainz, 2017-06-02:
@Oros42 ¿Is this find_cell_id utility usefull, or it displays what already simple-imsi-catcher do?

Oros42, 2017-06-03:
It's an old code who was merged in simple_IMSI-catcher.py.

petterreinholdtsen, 2017-09-29:
Perhaps time to remove it from the repo if it is no longer relevant?

commit 808ff3ee (2017-10-02): close #9
closed 2017-10-02 by Oros42