Oros42/IMSI-catcher #10: IMSI Periodic

engrmasood2002, 2017-06-23:
Hi, I only see IMSI once when i power up mobile in front of RTL-SDR dongle. 
Why IMSI not shown periodically?. 
Does mobile send it periodically?

Oros42, 2017-06-25:
This is what have understand of GSM :
When a phone come to the new area of an antena, then it send the IMSI and get a TMSI.
The phone use the TMSI with this antenna.
But I don't know why some phones of «Bouygues Telecom» then periodically their IMSI.

engrmasood2002, 2017-06-25:
Thanks for the reply, how can i get IMSI from TMSI because i dont get IMSI of phones already connected with mobile system antenna. can you provide any code example. Thanks

Oros42, 2017-06-25:
You can't get IMSI form TMSI already connected. TMSI is a random number.

engrmasood2002, 2017-06-25:
Ok. It would be great if you provide answer to these questions as I know c/c++ but not python and scripts are written in python.
1. Why the warning "no route to destination"  comes  when running imsi catcher script.?
2. Does imsi catcher script output imsi packets to any port (udp or TCP) ?

Oros42, 2017-06-25:
1. "no route to destination" is a waring from  the scapy library who try to use ipv6. It's not important.
2. my IMSI-catcher doesn't output on the network ! But listen packet from gr-gsm / airprobe_rtlsdr.py or any GSM decoder who send data to the (local) network.

engrmasood2002, 2017-06-26:
Just wondering why not only one imsi instead of imsi1 and imsi2 in source code.

engrmasood2002, 2017-06-26:
If I want to receive imsi of phone connected with different mobile operator then I think connecting more dongle can be a solution. Could multiple instances airprobe script can be run? If yes then how force multiple airprobe scripts to attach to different dongles in a computer.

Oros42, 2017-06-26:
I don't know. Ask it to the author of airpobe https://github.com/ptrkrysik/gr-gsm

engrmasood2002, 2017-06-27:
Thanks for the responses, 
Wondering if just want to receive imsi from one mobile operator network phones. Do gsm operator have single uplink frequency or multiple. If multiple then are they under RTL dongle bandwidth or we have to retune even for single mobile operator phones.

engrmasood2002, 2017-06-28:
Just wondering why not only one imsi instead of imsi1 and imsi2 in source code.  Whats different between IMSI1 and IMSI2?

Oros42, 2017-06-28:
Please read links at the bottom of the README.md

engrmasood2002, 2017-06-29:
Thanks alot for the responses. 
Is there any way to get MSISDN (user phone no) from IMSI ?

Oros42, 2017-06-29:
No
IMSI is link to the operator and not to the phone.

engrmasood2002, 2017-06-29:
Is there any scenario where you have seen last ten digits of MSISDN same as last ten digits of IMSI. According to wiki the last section of both the MSISDN and IMSI are ten bytes each. I mean have you seen your phone no. Dogits same in imsi upto any number of fields.

Oros42, 2017-06-29:
In France, MSISDN and IMSI are totally different. You can change your operator and keep the same MSISDN for new IMSI.

engrmasood2002, 2017-06-30:
Is imsi sent in gsm frequencies 800-900Mhz if phone is connected with 3G or 4G or we need to tune in 3G or 4G band?

Oros42, 2017-07-06:
It's depend of your country and operator.
You could have one of this frequencies : 700 Mhz, 800 Mhz, 900 Mhz, 1800 Mhz, 2100 MHz,  2600 MHz.

engrmasood2002, 2017-07-07:
What channel in gsm used to send IMSI? is it SDCCH or else ?

Oros42, 2017-07-08:
It's BCCH with message type: Paging Request Type 1.
https://www.ckn.io/blog/2015/11/29/gsm-sniffing-sms-traffic/

engrmasood2002, 2017-07-09:
Thanks for your kind replies. I was just thinking  that your imsi catcher gets imsi from downlink frequencies and we may get imsi of phones that are many miles away. Is there any channel which phone use to send imsi(on uplink frequency) ?

Oros42, 2017-07-11:
For listening uplink, you should add -45Mhz offset.
So 928.8 Mhz in down = 882.8 Mhz in up.
I haven't try (yet) to catch uplink packet with wireshark. But if you do it, send me the network capture and I will upgrade my program.

engrmasood2002, 2017-07-12:
I tuned to uplink frequency of my area but there is no output from imsi catcher script. I don't know whether airprobe_rtlsdr.py supports uplink decoding or not or imsi catcher script needs to be modified.

Oros42, 2017-07-12:
For checking if airprobe_rtlsdr.py works on uplink, run wireshark :
```
sudo wireshark -k -Y '!icmp && gsmtap' -i lo
```
Also, you can user gqrx to see if you have uplink.
```
sudo apt install gqrx-sdr
gqrx
```

engrmasood2002, 2017-07-12:
I have already run wireshark but no output from airprobe_rtlsdr.py for uplink frequencies. May be gqrx decodes uplink. i will check it.

Oros42, 2017-07-13:
If airprobe_rtlsdr.py can't decode uplink, I can't do anything :-/

engrmasood2002, 2017-07-13:
Thanks for providing useful information throughout this session.

closed 2017-07-26 by Oros42