Oros42/IMSI-catcher #11: 404 Failed to fetch http://ppa.launchpad.net/ptrkrysik/gr-gsm/ubuntu/dists/trusty/main/binary-amd64/Packages

nachimehta, 2017-07-08:
after running the commands I get the following error:

`Failed to fetch http://ppa.launchpad.net/ptrkrysik/gr-gsm/ubuntu/dists/trusty/main/binary-amd64/Packages`

Seems the package has been moved? Any updates on the new location?

nachimehta, 2017-07-08:
http://blog.nikseetharaman.com/gsm-network-characterization-using-software-defined-radio/ also leads to a dead end now.

Oros42, 2017-07-09:
Ubuntu Trusty is realy old. Upgrade your OS to the last version.
Or you can try to change
```
http://ppa.launchpad.net/ptrkrysik/gr-gsm/ubuntu/dists/trusty/main/binary-amd64/Packages
```
to
```
http://ppa.launchpad.net/ptrkrysik/gr-gsm/ubuntu/dists/xenial/main/binary-amd64/Packages
```

yerpj, 2017-07-26:
Same problem for me.
@Oros42 Could you please specify where to change this URL?
Thank you!

Oros42, 2017-07-26:
You could try this :
```
apt-get install software-properties-common dirmngr
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys CD383E1262BECEC3
apt-add-repository 'deb http://ppa.launchpad.net/ptrkrysik/gr-gsm/ubuntu xenial main'
apt update
apt install gr-gsm python-numpy python-scipy python-scapy
```
But don't work at the end on Debian 9 :-/

Oros42, 2017-07-26:
Other solution for gr-gsm -> https://github.com/ptrkrysik/gr-gsm/wiki/Installation

yerpj, 2017-07-27:
Thanks! 
Meanwhile I upgraded my Ubuntu to 16.04 LTS and the standard procedure hereafter worked like a charm.
```
sudo apt install python-numpy python-scipy python-scapy

sudo add-apt-repository -y ppa:ptrkrysik/gr-gsm
sudo apt update
sudo apt install gr-gsm
```

petterreinholdtsen, 2017-09-29:
The gr-gsm package is available directly from Debian now, see https://tracker.debian.org/pkg/gr-gsm .  It is not yet in Ubuntu, but should show up there after the next Ubuntu package sync with Debian.

closed 2019-01-20 by Oros42