Oros42/IMSI-catcher #12: No o/p in hackrf

beeprime, 2017-08-01:
Hi, I am using hackrf sdr and upon firing simple_imsi_catcher.py - it gives nothing apart from the first line saying cell_id, country etc...

BlackPhreaker, 2017-08-01:
Good afternoon and through wireshark data are viewed ???

beeprime, 2017-08-01:
No. Wireshark is up and running but no data are viewed.

BlackPhreaker, 2017-08-01:
Not in a script not in Wireshark data does not arrive ??? And the frequency of the GSM channel is configured to rightly try to scan the nearest BTS???

BlackPhreaker, 2017-08-01:
Example kal -s GSM900 or gsm_scanner.py Data is given or not ??

Oros42, 2017-08-01:
@beeprime no data in wireshark == wrong frequency

BlackPhreaker, 2017-08-02:
@beeprime To determine the channel and frequency of BTS, use the scanner from the package gr-gsm (gsm_scanner.py -v) !!!

beeprime, 2017-08-02:
While using RTL2832U in place of Hackrf One, I am getting IMSI numbers. That implies that simple_imsi_catcher is working fine with RTL-SDR but no GSM data is shown in wireshark while it's visible by running airprobe_rtlsdr.py from another console which is other that from where simple_imsi_catcher is running. Please help... (firstly wireshark gives LUA error but after resolving also no data is shown)

BlackPhreaker, 2017-08-02:
@beeprime * kal -s GSM 900 displays the result if there is no reinstall the hackrf One software. I have everything in one system working normally RTL2832U, Hackrf One, OsmocomBB... For the sake of interest probyval radio takes ??

Oros42, 2017-08-02:
@beeprime if you have nothing in wireshark, the problem didn't come from my IMSI-catcher.
So check : 
- your hackrf is correctly plug
- you have the last firmware of hackrf
- you have the last gr-gsm
- run gqrx (sudo apt install gqrx-sdr) and check you get signal of cell tower

closed 2019-01-20 by Oros42