Oros42/IMSI-catcher #14: GPS coordinates !!!

BlackPhreaker, 2017-08-02:
@Oros42 Do you think it would be useful for you to display GPS coordinates in a script from opencellid.org ??? And is it difficult to do this ???

Oros42, 2017-08-02:
The precision of opencellid.org is not very good. And can't locate some cell towers in my town.
That why, I haven't add GPS coordinates.

BlackPhreaker, 2017-08-02:
Yes you're right the accuracy is really bad ... And what additions and changes do you plan in the future ??? And how do you feel about additional ideas ???

E3V3A, 2017-08-02:
@BlackPhreaker Why don't you post your own ideas and modifications instead of asking everyone else for it? Looking at your repo, there is nothing, but you constantly ask for help and conclude you will fix and modify yourself. Please share!

BlackPhreaker, 2017-08-03:
@E3V3A E:V:A Grieve you but my developments will not be laid out in open access ... At least a large part of them))

E3V3A, 2017-08-17:
@Oros42 I suggest you change your CC0 license to GPLv2 or something similar to keep people like @BlackPhreaker from stealing your work and then put it under their own Copyright. It's sad to see people abusing others for their own purposes. Clearly BlackPhreaker have no intention in sharing anything, nor helping resolve any issues and only make irrational demands.

Oros42, 2017-08-17:
I don't care if someone want to be an asshole. It's his own problem. Not mine.
Changing the license will not stop this type of person so I keep this license.
For me every thing should be in public domain.

BlackPhreaker, 2017-08-18:
@E3V3A , @Oros42 !!! Do not throw such loud words!!! What's wrong if a person changes freely laid out code for their needs ???

petterreinholdtsen, 2017-09-10:
What are the best sources for GPS location of base stations?  I've come across these, but the accuracy is very variable:
 * http://opencellid.org/
 * http://cellidfinder.com/
 * https://location.services.mozilla.com/
 * https://www.cellmapper.net/map

Which source would you recommend for getting the GPS coordinate?

BlackPhreaker, 2017-09-10:
@petterreinholdtsen For my country, the best result was: 
http://opencellid.org/
http://find-cell.mylnikov.org/

closed 2020-04-18 by Oros42