Oros42/IMSI-catcher #16: pybombs installation

ikechits, 2017-08-16:
hi, I am new to Pybombs and im trying to make an msi catcher as a subsysytem for a bigger project on Ubuntu, however I'm running into problems installing Pybombs. I'm still learning python and I had problem installing python as well the I'm having problems with the recipes and I tried and alternative route to install it which as the one on your website and it seems to be the soapy files and i get this out put please help or point me in the right direction

k9@k9-HP-Notebook:~$ sudo pybombs install gr-gsm
PyBOMBS - INFO - PyBOMBS Version 2.3.1
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
The directory '/home/k9/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Install tree:
|
\- gr-gsm
   |
   \- gr-osmosdr
      |
      +- airspy
      |
      +- hackrf
      |
      \- soapysdr
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: soapysdr
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/soapysdr/build
Configuring: (100%) [=========================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- SWIG_FOUND: TRUE - 2.0.12
-- PYTHONINTERP_FOUND: TRUE - 2.7.12
-- PYTHON_EXECUTABLE: /usr/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python2.7/dist-packages
-- PYTHONLIBS_FOUND: TRUE - 2.7.12
-- PYTHON_INCLUDE_DIRS: /usr/include/python2.7
-- PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython2.7.so
-- CMAKE_SWIG_FLAGS=-c++;-threads
-- 
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 2.0.12
-- Could NOT find Python3InterpDbg (missing:  PYTHON3_DBG_EXECUTABLE) 
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/local/bin/python3
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.4/distutils/sysconfig.py", line 13, in <module>
    import re
  File "/usr/local/lib/python3.4/re.py", line 336, in <module>
    import copyreg
  File "/usr/local/lib/python2.7/dist-packages/copyreg/__init__.py", line 7, in <module>
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.
-- PYTHON3_INSTALL_DIR: ${prefix}/
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/local/include/python3.4m
-- PYTHON3_LIBRARIES: /usr/local/lib/libpython3.4m.a
-- CMAKE_SWIG_FLAGS=-c++;-threads
CMake Error at python3/CMakeLists.txt:87 (install):
  install TARGETS given no LIBRARY DESTINATION for module target
  "_SoapySDR3".

CMake Error at python3/CMakeLists.txt:92 (install):
  install FILES given no DESTINATION!

-- 
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
-- 
 * Library , runtime library v0.6.1-g285e72aa
 * Tests , library unit tests
 * Docs , doxygen documentation
 * Python , python bindings v2.7.12
 * Python3 , python3 bindings

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
-- 

-- SoapySDR version: v0.6.1-g285e72aa
-- ABI/so version: v0.6
-- Install prefix: /usr/local
-- Configuring incomplete, errors occurred!
See also "/usr/local/src/soapysdr/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/soapysdr/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package soapysdr:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package soapysdr. Aborting.

E3V3A, 2017-08-17:
@ikechits  Please use triple `back-ticks` (look like back slanting single quotes, i.e ``` ` ```) to enclose pasted code or better, use a pastebin link. It's nearly impossible to read what you have pasted.

ikechits, 2017-08-17:
sorry, thank you for the timely response! and done!

Oros42, 2017-08-17:
Have you try first this setup :
```
sudo add-apt-repository -y ppa:ptrkrysik/gr-gsm
sudo apt update
sudo apt install gr-gsm
```
I'm not the author of gr-gsm and I can't resolve your problem. So you can go here : https://github.com/ptrkrysik/gr-gsm/issues

petterreinholdtsen, 2017-08-27:
I'm working on getting gr-gsm into Debian, see https://bugs.debian.org/871055 for the status.  The pybombs approach is sensitive to version skew and incompatible libraries.  I ran into problems when I tried it, as openssl had changed.

closed 2019-01-20 by Oros42