Oros42/IMSI-catcher #17: Run without root privileges?

petterreinholdtsen, 2017-08-27:
With the introduction of the gr-gsm change in https://github.com/ptrkrysik/gr-gsm/pull/317 it is possible to set up a non-privileged UDP collector service to gather the GSM data, instead of using wireshark/scapy.

What about rewriting simple_IMSI-catcher.py to listen on the appropriate port?  Something like this patch should do it:

```
diff --git a/simple_IMSI-catcher.py b/simple_IMSI-catcher.py
index 5e6c43e..ea133df 100644
--- a/simple_IMSI-catcher.py
+++ b/simple_IMSI-catcher.py
@@ -63,6 +63,8 @@ Realtek RTL2832U : http://doc.ubuntu-fr.org/rtl2832u and http://doc.ubuntu-fr.or
 from scapy.all import sniff
 import json
 from optparse import OptionParser
+import socket
+import sys
 
 imsis=[] # [IMSI,...]
 tmsis={} # {TMSI:IMSI,...}
@@ -400,6 +402,17 @@ def find_imsi(x):
 				imsi2=p[0x48:][:8]
 				show_imsi(imsi1, imsi2, tmsi1, tmsi2, p)
 
+def udpserver(port, prn):
+        sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
+        server_address = ('localhost', port)
+        #print >>sys.stderr, 'starting up on %s port %s' % server_address
+        sock.bind(server_address)
+        print >>sys.stderr, '\nwaiting to receive message'
+        while True:
+                data, address = sock.recvfrom(4096)
+                # Pad packages as a workaround for find_imsi()
+                # expecting raw packages, not just the UDP data.
+                prn('x' * (81 - 39) + data)
 
 if __name__ == '__main__':
 	parser = OptionParser(usage="%prog: [options]")
@@ -407,6 +420,7 @@ if __name__ == '__main__':
 	parser.add_option("-i", "--iface", dest="iface", default="lo", help="Interface (default : lo)")
 	parser.add_option("-m", "--imsi", dest="imsi", default="", type="string", help='IMSI to track (default : None, Example: 123456789101112 or "123 45 6789101112")')
 	parser.add_option("-p", "--port", dest="port", default="4729", type="int", help="Port (default : 4729)")
+	parser.add_option("-s", "--sniff", action="store_true", dest="sniff", help="sniff on interface instead of listening on port")
 	(options, args) = parser.parse_args()
 
 	show_all_tmsi=options.show_all_tmsi
@@ -435,4 +449,7 @@ if __name__ == '__main__':
 		mcc_codes = json.load(file)
 
 	print("{:7s} ; {:10s} ; {:10s} ; {:17s} ; {:12s} ; {:10s} ; {:21s} ; {:5s} ; {:4s} ; {:5s} ; {:6s}".format("Nb IMSI", "TMSI-1", "TMSI-2", "IMSI", "country", "brand", "operator", "MCC", "MNC", "LAC", "CellId"))
-	sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi, store=0)
+	if options.sniff:
+		sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi, store=0)
+	else:
+		udpserver(port=options.port, prn=find_imsi)
```

Oros42, 2017-08-28:
Mmmh good :-)
I haven't test yet but I think I will add this to my code.
Thanks you

petterreinholdtsen, 2017-09-04:
This was fixed in commit 977fed1c4d5e5312f1c1f3df5fd2445ddeb0dd06.

closed 2017-09-04 by petterreinholdtsen