Oros42/IMSI-catcher #27: Tabs or spaces? (python 3 related)

petterreinholdtsen, 2017-09-07:
Hi.  Python 3 is more picky than python 2 when it comes to indenting with space or tabs.  And the code currently only work with python 2, partly because it mixes tabs and spaces.  Do you have any opinion if tabs or spaces should be used?  Personally I prefer spaces, for various reasons, but my preference is not very strong.

Can we pick tabs or spaces, document the preference and convert the code to use it?  Switching to python 3 would make several charset encoding issues easier to solve, and I believe it is a good long term goal.

Oros42, 2017-09-07:
Please use tabs !

Tabs are better.
With tabs :
- you have less characters
- in your editor, you can chose the size of the indentation (ie: tabs = 2 spaces or 4 spaces or...)

petterreinholdtsen, 2017-09-07:
[Oros42]
> Please use tabs !

OK.  Can you tabify the source to ensure consistency, and increase the
chance of running it in python 3?

> With tabs :
> - you have less characters

True.  I would be surprised if it have any effect on any measurable
metrix, though. :)

> - in your editor, you can chose the size of the indentation (ie: tabs
> = 2 spaces or 4 spaces or...)

Not really true, if you like me sometimes want things to align
horizontally involving something that is not a tab, like this example
with the function arguments:

  def myfunc(a,
	     b,
	     c,
	     d)

Anyway, I do not see it as my misssion to convince you that space are
better than tabs, so tabs it is. :)

-- 
Happy hacking
Petter Reinholdtsen

closed 2017-09-10 by petterreinholdtsen