Oros42/IMSI-catcher #28: socket.error: [Errno 98] Address already in use

Vali9101, 2017-09-07:
Hi!

I'm trying to get the script running but after getting gr-gsm running, when starting the IMSI-catcher I get the following error:

root@kali:~/IMSI-catcher# python simple_IMSI-catcher.py
Saving to SQLite database in imsi-catcher.sqlite
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC  ; MNC   ; LAC    ; CellId
Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 534, in <module>
    udpserver(port=options.port, prn=find_imsi)
  File "simple_IMSI-catcher.py", line 487, in udpserver
    sock.bind(server_address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use
root@kali:~/IMSI-catcher# 

Does any one have an idea for me or know of this issue? Ive scoured muliple forums but can't seem to find something that works.

Thanks so much in advance.

Oros42, 2017-09-07:
Use ```python simple_IMSI-catcher.py --sniff```

Vali9101, 2017-09-08:
thank you so much @Oros42 ! Highly appreciated. problem solved.

closed 2017-09-08 by Oros42