Oros42/IMSI-catcher #32: TypeError: ouput() takes at most 12 arguments (13 given)

ajayfuloria, 2017-09-24:
Hi !

I am trying to use simple_IMSI-catcher.py but am facing the following problem. Seems output function is being passed more arguments than defined. But in the code I found output being passed 12 args only.

Pls help. I am using master branch of grgsm using hackrf for receiving. 

thanks

merlin@den:~/work/gsm/IMSI-catcher$ sudo python simple_IMSI-catcher.py --sniff
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC  ; MNC   ; LAC    ; CellId
Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 577, in <module>
    sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi_from_pkg, store=0)
  File "/usr/local/lib/python2.7/dist-packages/scapy/sendrecv.py", line 620, in sniff
    r = prn(p)
  File "simple_IMSI-catcher.py", line 539, in find_imsi_from_pkg
    find_imsi(udpdata)
  File "simple_IMSI-catcher.py", line 470, in find_imsi
    t.register_imsi(gsm.arfcn, imsi1, imsi2, tmsi1, tmsi2, p)
  File "simple_IMSI-catcher.py", line 278, in register_imsi
    self.pfields(str(n), self.str_tmsi(tmsi1), self.str_tmsi(tmsi2), imsi1, str(self.mcc), str(self.mnc), str(self.lac), str(self.cell), p)
  File "simple_IMSI-catcher.py", line 214, in pfields
    self.ouput_function(cpt, tmsi1, tmsi2, imsi, imsibrand, imsicountry, imsioperator, mcc, mnc, lac, cell, packet)
TypeError: ouput() takes at most 12 arguments (13 given)

Oros42, 2017-09-24:
Mmmh commit 9ff85f9d9212b4ab01dfe34b13e842b43571a7fa didn't fix it ? 0_o

ajayfuloria, 2017-09-24:
Thanks, 

The issue is fixed in the commit you mentioned. 

Thanks

closed 2017-09-24 by Oros42