Oros42/IMSI-catcher #35: No funciona

owl-city, 2017-10-14:
Hola, hace unos días instalé Ubutu 16.04 LTS y con el todas las dependencias y herramientas que se necesitan para poder utilizar GNURadio junto con un stick de radio RTL-SDR.
Al momento de utilizar  gr-gsm livemon me arroja un error en simple_IMSI-catcher.py:

```
root@Scott:/home/scott/IMSI-catcher# python simple_IMSI-catcher.py
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC  ; MNC   ; LAC    ; CellId
Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 582, in <module>
    udpserver(port=options.port, prn=find_imsi)
  File "simple_IMSI-catcher.py", line 531, in udpserver
    sock.bind(server_address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use

Y sucesivamente alrevés si ejecuto primero  simple_IMSI-catcher.py y después gr-gsm livemon me arroja lo siguiente:

Executing: /usr/bin/python2 -u /home/scott/pybombs/src/gr-gsm/apps/grgsm_livemon.py
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-295-ga0adcd33
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
[32;1m[INFO] [UHDlinux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-191-g1cd96dde] [39;0m
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
No supported tuner found
Enabled direct sampling mode, input 1
Exact sample rate is: 2000000.052982 Hz
Traceback (most recent call last):
  File "/home/scott/pybombs/src/gr-gsm/apps/grgsm_livemon.py", line 357, in <module>
    main()
  File "/home/scott/pybombs/src/gr-gsm/apps/grgsm_livemon.py", line 345, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/home/scott/pybombs/src/gr-gsm/apps/grgsm_livemon.py", line 176, in __init__
    self.blocks_socket_pdu_0_0 = blocks.socket_pdu("UDP_SERVER", '127.0.0.1', serverport, 10000, False)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig5.py", line 419, in make
    return _blocks_swig5.socket_pdu_make(*args, **kwargs)
RuntimeError: bind: La dirección ya se está usando
>>> Done

```
He instalado Kali linux, Ubuntu GNURADIO enviroment live y ahora estoy usando Ubuntu y no me funciona correctamente, estoy llegando a pensar que es mi stick de radio el que no funciona porque ya he instalado correctamente todo y he puesto los blacklist del RTL como corresponde.
El stick de radio es un RTL2832U + FC0012 Mini DVB-T + DAB + + FM USB Digital TV Dongle-Negro, lo compré aquí : [http://www.dx.com/es/p/rtl2832u-r820t-mini-dvb-t-dab-fm-usb-digital-tv-dongle-black-170541?tc=MXN&gclid=EAIaIQobChMIstLBwLXv1gIVQ0AbCh0sSwHGEAYYAiABEgIxifD_BwE#.WeGllHBrw8o](url)  

Agradecería mucho que me orientaran sobre éste problema, sin más un saludo.

Oros42, 2017-10-15:
Same issue as #28 
Use ```python simple_IMSI-catcher.py --sniff```

Oros42, 2017-10-18:
If you have nothing in simple_IMSI-catcher.py, change the frequency in grgsm_livemon

owl-city, 2017-10-19:
Ya lo hice, mi frecuencia es 951300000 y lo he analizado con Kalibrate-rtl, he puesto kal -s GSM900 y me a arrojado esa frecuenia pero no me captura nada en simple_IMSI-catcher.py, el problema que acabo de darme cuenta es que tengo un SDR-RTL con un chip Fitipower FC0012 (22 - 948.6 MHz) y las capturas son con un chip Rafael Micro R820T (24 - 1766 MHz) supongo que no me funciona por eso.

```
root@Scott:/home/scott# kal -s GSM900
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
No supported tuner found
Enabled direct sampling mode, input 1
Exact sample rate is: 270833.002142 Hz
kal: Scanning for GSM-900 base stations.
GSM-900:
	chan: 82 (951.4MHz + 15.633kHz)	power: 29471.15

```
Sin embargo bajo la frecuencia de 951.4 a 951.3 porque no me captura tráfico.
https://i.imgur.com/7OlVnL7.jpg

No captura de manera correcta el tráfico, por lo que la antena no funciona correctamente, lo que he hecho fue adquirir una nueva antena con chip Rafael Micro R820T y en cuanto llegue probaré para ver si funciona. Si hay alguna manera de hacer que me funcione mi antena y poder capturar tráfico con el script IMSI-simple te lo agradecería mucho.

Oros42, 2017-10-21:
Okay, so your problem is in your antenna/RTL-SDR and not in my simple_IMSI-catcher.py.

FFY00, 2017-10-21:
You probably need to adjust the ppm error to address the frequency shift (I suppose you are saying that in 951.4 there's traffic but in 951.3 there isn't).

closed 2019-01-20 by Oros42