Oros42/IMSI-catcher #38: I can find my own MS.

eveding, 2017-10-23:
Good evening. 

I was doing a IMSI-CATCHER and i get it. It Works perfect, i found a loot of MS and they information, but i cant find a my own imsi number. I made a lot test, putting my phone into GMS and trying to conect to the same BTS and nothing.

I'm using next ones

RTL2832U donggle
E4000 donggle

Core i5 8gb Ram
Ubuntu 16-04

Oros42, 2017-10-24:
Not easy. I'm trying to write an help for that. But not finish.

I use Minicom (https://alioth.debian.org/projects/minicom) to try to understand what does my phone.
https://ccdcoe.org/cycon/2015/proceedings/16_xenakis_ntantogian.pdf (page 12)

Y2Kot, 2017-12-15:
I'm trying the same:
0) I have 2 phones on one arfcn/freq and I know both imsi
1) I know ARCN/Freq of my devices
2) I'm scanning this freq/arfcn with my bladerf
3) trying it more then 3 hours long
3.1) send sms and make calls
4) Can't find none of 2 mobile phones???
Can you give me explanation of my mistakes?

kernelfg, 2018-01-24:
I tried the same with 4 different mobile phone, without any success. None of them are showing. I don't see what I'm doing wrong.
Thanks

DerLomo, 2024-10-30:
> I tried the same with 4 different mobile phone, without any success. None of them are showing. I don't see what I'm doing wrong. Thanks

did you find any solution?