Oros42/IMSI-catcher #41: simple_IMSI-catcher look like to work but not scan-and-livemon.

sckebby, 2018-01-11:
Hi Oros42, 

I am french, but I try to write you in english for the other people followings your project.
Compliment for the job you have done, I am facing a strange issue, without apparently errors.

In the First terminale I start:

**simple_IMSI-catcher.py** look like to work but not **scan-and-livemon**.

so when I start "**scan-and-livemon**" th result I have look like to be empty, and stop quickly.

Locating potential GSM base station frequencies (this can take a few minutes).
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

I am using it on Raspberry Updated.
I have following all different test to install gr-gsm, the grgsm_scanner produce the same output.

Locating potential GSM base station frequencies (this can take a few minutes).
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

My USB DVB-T is configured right, when I do:

**kal -s GSM900**  ( To scan channel )

Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 270833.002142 Hz
kal: Scanning for GSM-900 base stations.
GSM-900:
	chan: 23 (939.6MHz + 24.800kHz)	power: 27664.59
	chan: 53 (945.6MHz + 25.843kHz)	power: 99831.14

After I do:

**kal -c 53**
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 270833.002142 Hz
kal: Calculating clock frequency offset.
Using GSM-900 channel 53 (945.6MHz)
average		[min, max]	(range, stddev)
+ 28.251kHz		[24167, 28987]	(4821, 1024.137451)
overruns: 0
not found: 0
average absolute error: -29.876 ppm

Where can I have a look for to resolve this issue.

Without errors, it's difficult to me to understand where I get wrong.

Thanks for support, 

Regards,

Christian

Keskebeu, 2018-03-07:
That's a gr-gsm question.

Keskebeu, 2018-03-07:
Sorry a kalibrate-rt question

Keskebeu, 2018-03-07:
Its not an issue the c option tells you but how much your clock is offset to the selected channel. And that's equal to:
average absolute error: -29.876 ppm

closed 2019-01-20 by Oros42