Oros42/IMSI-catcher #42: imsi catcher.py wont run simultaneously with scan-and-livemon

aoyue2702, 2018-01-18:
I am unable to get IMSI-catcher.py to run simultaneously with scan-and-livemon.

I am on ubuntu and I have tried both installation methods you recommend.

`sudo python scan-and-livemon
[sudo] password for dog: 
Locating potential GSM base station frequencies (this can take a few minutes).
Traceback (most recent call last):
  File "scan-and-livemon", line 101, in <module>
    main()
  File "scan-and-livemon", line 78, in main
    list = find_gsm_bases()
  File "scan-and-livemon", line 25, in find_gsm_bases
    (options, args) = scanner.argument_parser().parse_args()
AttributeError: 'module' object has no attribute 'argument_parser'

``sudo python simple_IMSI-catcher.py
sudo: unable to resolve host dog1853
[sudo] password for dog: 
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC  ; MNC   ; LAC    ; CellId

Keskebeu, 2018-01-22:
Thats because you have to black list the rtl you are using. If you have used in a different OS like Windows than its meant to use with windows.

So blacklist this device and it should work. 
+ another problem is that grgsm does not support frequencies higher than PCS1900 and lower than 800 meaning that anything above that like 2000mhz wont be supported. I have been using this software and its fairly easy to get imsis and everything else but everything has to be 2g otherwise nothing will show up.

This is meant for 2g frequencies really old by now. What you should is connect your phone to gsm only on android (so you dont have to cause trouble by jamming) and than you will see what is happenning.

Let me know

Mortensv, 2018-03-07:
do you have to bkacklist it in windows or linux?

Keskebeu, 2018-03-07:
Blacklist it on linux.

FFY00, 2018-03-07:
```
sudo modprobe -r dvb_usb_rtl28xxu
```

Should do it.

Keskebeu, 2018-03-12:
When ppl is not locked than you dont have any frequencies nearby. Or if you do, you have to play around with ppm and gain options. 

I could not pick up signal with rtl-sdr even by trying to set a ppm and gain, but with ettus b200 I was picking up everything. So check it out.

closed 2019-01-20 by Oros42