Oros42/IMSI-catcher #43: IMSI not showing up

kernelfg, 2018-01-24:
Hi,

What could be the reason why my own IMSI is not showing up for every tests I did ?
Thanks in advance

alexandr84, 2018-02-01:
I'd like to know this as well. I find a LOT of IMSI numbers, but not my own IMSI.

imranrajjad, 2018-02-07:
well same here.. unable to find my own.. do we need to process the found IMIS`s further?

imranrajjad, 2018-02-08:
well I had Wireshark open next to the IMSI catcher script. So it appears that for  `Message Type: Location Updating Request (0x08)` also carries IMSI which are not being sniffed by script

imranrajjad, 2019-01-21:
I checked ARFCN from phone`s service settings, its the same

asiudoisjdfo, 2020-02-20:
Seems that this is an issue for multiple people. I've had issues also. Any news on this?

brianblank, 2023-03-12:
According to https://harrisonsand.com/posts/imsi-catcher/ , the IMSI is only transmitted during initial connection which negotiates a Temporary Mobile Subscriber Identity (TMSI).  All subsequent communications happens with the TMSI.  I'm curious if this is the issue you are seeing.  As this is a passive connection and not an active connection, I don't think you can force the phone to reveal its IMSI.

To test this out, try turning your phone off and on again to force it to establish a new initial connection and see if your TMSI shows up.

Does this IMSI capture tool also capture the TMSI too?  If so, I'm sure the phone company can map the TMSI back to an IMSI with a valid search warrant.

DerLomo, 2024-08-02:
> To test this out, try turning your phone off and on again to force it to establish a new initial connection and see if your TMSI shows up.

Didnt work for me