Oros42/IMSI-catcher #45: About how it works TX RX

Micolocobr2, 2018-02-14:
Hi,

Does this IMSI catcher works just receiving celular data from DVB-T? I mean DVB-T does not have to transmit data to cell phone in order to this cell phone register on the IMSI catcher as cell phones normaly does with regular cell towers?

Oros42, 2018-02-17:
Yes, this IMSI catcher only receive celular data by listening cell towers. There is no TX.

Micolocobr2, 2018-02-17:
I thought IMSI catcher listen to cell phones, I was wrong... So in order to catch many cell numbers I dont need to run over all places searching to the numbers. I just need to stand next to cell towers and wait to signals to catch. Is it correct?

Oros42, 2018-02-17:
My IMSI catcher only listen cell towers. But it's not the case of all IMSI catcher.
Yes, you don't need to run over all places and you could be at few kilometers of the cell tower.

closed 2019-01-20 by Oros42