Oros42/IMSI-catcher #46: Its Alivveeee!!!!

Micolocobr2, 2018-03-06:
Thanks Oros42, FFY00 and other guys for all the help.

My IMSI Catcher is on line and imsi numbers are now popping up over here.

Is there a way to capture the cell numbers of those imsi codes?

Oros42, 2018-03-06:
In most cases, you can't. The cell number is not send.
And only in few countries you have imsi == cell numbers.
:-/

Micolocobr2, 2018-03-07:
Is there some peace of software I can use to get the numbers?

FFY00, 2018-03-07:
No. Only the operators know that.

Micolocobr2, 2018-03-09:
Is there some imsi catcher that listen to the phones and not only to the cell towers?

Oros42, 2018-03-09:
Yes
With ```grgsm_livemon``` and ```wireshark``` you could listen cell phones. But you have to find the good frequency.