Oros42/IMSI-catcher #47: IMSI Catcher not showing imsi etc

dave-88, 2018-03-25:
Dear Community,
Can anybody please help me.
Install all software to run my program.
But it could not detect imsi and phone information.
Hope u know the problem
I could sniff only one thing.
If u help me personally I would also pay for help if it works great in the end like an professional one.
kindly
[image]

Oros42, 2018-03-26:
Have you try to listen different frequencies ?

dave-88, 2018-03-28:
Dear Oros,

I did,nothing happend for real:(

Do you have any further infos ?

kindly

 

Sent: Monday, March 26, 2018 at 11:06 PM

dave-88, 2018-03-29:
If u can help me personally I would also donate to you .
Please let me know.

Oros42, 2018-04-02:
Hello, sorry for the delay (a lot of work).

Which range of frequencies did you use ?
How many frequencies did you try ?
How long did you stay on the same frequency ?
Do you have any traffic in grgsm_livemon ?
Did you get MCC, MNC and LAC ?

dave-88, 2018-04-03:
I use all frequencies for testing I had in  my window.
I was a few seconds on each frequencies waiting then i used next one.
I didnt got any informations about MCC MNC and LAC.
What u mean which traffic exactly?In my diagramm there was a "blue wave" which went up and down.
If possible can I hit u up here personaly via PM ?
Its very important for my work,also I can donate in cryptos .
kindly

Oros42, 2018-04-03:
Okay
Send me an email at : imsi at ecirtam.net