Oros42/IMSI-catcher #48: Why I can't find IMSI number for my own phone?

zhongrui960326, 2018-04-04:
Hi,

I was using this project to find some IMSI numbers with USRP N210 on the kali , and I really get a lot ,but I can't find  my own imsi number. I made my phone work on the GSM and  tried to send sms or make some calls ,but I still can't find it .

Can you give me explanation of my mistakes?
Thanks!

FFY00, 2018-04-04:
Before you try something like this you should take a look at the protocol. The IMSI is rarely sent, you probably will only send the IMSI over the air when the phone first connects to the basestation. Also, you need to make sure you're listening tye right frequency.

ciphercc, 2019-04-28:
Hi,

Can you provide script for active mode imsi catcher and 
what hardware require for active imsi catcher