Oros42/IMSI-catcher #49: Hello,pleases does this support 3G, 4G network?

xiaojiazu, 2018-04-23:


kolet, 2018-09-30:
> My project is base on [gr-gsm](https://github.com/ptrkrysik/gr-gsm). So it's only support GSM.
> But if you found a 3G or 4G decoder, feel free to share it :-)

isnt possible to create an imsi catcher for 3g / 4g ?

they are some countries that 2g are not popular anymore they are barely used.
most of the phones nowdays are on 3g/4g .
is that possible to catch their imsi, and when they do a call or getting an sms to forward to a real 3/4g network ?

ofc u need a device that support that frequency, i thought about LimeSDR . will that work?

Oros42, 2018-09-30:
I haven't found yet a good decoder like gr-gsm but for LTE :-/
I use an hackrf but LimeSDR is a good hardware.

kolet, 2018-10-01:
> I haven't found yet a good decoder like gr-gsm but for LTE :-/
> I use an hackrf but LimeSDR is a good hardware.

not looking to decode the date such sms or calls , not into it.
more like getting imeis and imsi of the handset that are connected to the BTS (rouge tower cell)

so my question was more like will all handset (cell phones) will connect to the BTS (imsi catcher/rouge tower cell) ?  have u asked ur self how does that work? how come the handset will switch to ur cell and not the ISP cell ?
if so do u mind explaining that to me? (or leaving me some IM contact info so we can live chat, will be more private and faster to communicate )

also, this is a good article 
https://pdfs.semanticscholar.org/63af/2b296e04f6e5ac51991bcdd04e06687623b2.pdf

Mera-balou, 2018-12-23:
Hi Oros42,

did you try theses open source LTE decoders?
- srsLTE : https://github.com/srsLTE/srsLTE
- gr-lte : https://github.com/kit-cel/gr-lte

Will you plan to support 3G, 4G network?

Thx

Oros42, 2019-01-12:
https://github.com/kit-cel/gr-lte seem interesting but there is not simple example to decode with an hackrf.

tomasdev, 2019-02-06:
https://www.shawngarringer.org/rtl-sdr-based-lte-signal-analyzer/
https://github.com/JiaoXianjun/LTE-Cell-Scanner

But I'm starting to think the RTL-SDR devices ($10) aren't capable of some frequencies. I'm in NYC and 0 detected after building properly.

closed 2020-04-18 by Oros42