Oros42/IMSI-catcher #51: Lots of IMSI-numbers from Bermuda and Guam, can this be correct?

sesom, 2018-05-03:
Hi,

this is not an issue, but a question. The scripts runs fine and after ~24 hours I already saw 35000+ distinct IMSI numbers. What puzzles me is the high number of numbers from Bermuda and Guam. The other numbers seem to be plausible, but I doubt that there where 1060 SIM-cards from the Bermudas around here at my place in Germany. Can it be, that some providers simply use these rather exotic countries like Bermuda and Guam, albeit the SIM-cards are for the German market?

```
sqlite> select count(distinct imsi) as count, imsicountry from observations group by imsicountry order by count desc limit 66;
count       imsicountry
----------  -----------
33067       Germany    
1060        Bermuda    
751         Poland     
464         Guam (Unite
271         Romania    
244         China      
164         Turkey     
150         Spain      
137         France     
116         Austria    
91          Guernsey (U
90          Italy      
89          Czech Repub
84          Bulgaria   
54          India      
53          Hungary    
48          Russian Fed
46          Croatia    
45          Netherlands
44          Ukraine    
40          Slovakia   
32          Serbia     
31          Egypt      
31          United Arab
29          Belgium    
27          Taiwan     
25          Greece     
24          Australia  
24          Switzerland
22          Lithuania  
21          Latvia     
21          Sweden     
20          South Korea
```

Oros42, 2018-05-03:
So you found the lost phones from Bermuda Triangle ? XD

You could try to update [IMSI-catcher/mcc-mnc/mcc_codes.json](https://github.com/Oros42/IMSI-catcher/blob/master/mcc-mnc/mcc_codes.json) with this :  
```
cd /mcc-mnc/
python update_codes.py
```

You can check if MCC and MNC you get are the same to [https://en.wikipedia.org/wiki/Mobile_Network_Code#Bermuda_-_BM](https://en.wikipedia.org/wiki/Mobile_Network_Code#Bermuda_-_BM)

sesom, 2018-05-04:
The lost phones from the Bermuda Triangle (and the lesser known Guam Square) seem all to be here in Bavaria. :)

I already had the newest mcc-mnc-version, but updated just to be on the safe side. The MCC-MNC-combinations I see are not listed on Wikipedia, but I can find them all at http://www.mcc-mnc.com/ ...

```
sqlite> select substr(distinct imsi,0,8) as imsistart, imsicountry, imsibrand, imsioperator from observations where imsicountry='Bermuda' group by imsistart order by imsi;
imsistart   imsicountry  imsibrand   imsioperator      
----------  -----------  ----------  ------------------
310 120     Bermuda      Sprint      Sprint Corporation
310 170     Bermuda      AT&T        AT&T Mobility     
310 260     Bermuda      T-Mobile U  T-Mobile USA      
310 410     Bermuda      AT&T        AT&T Mobility
```

Anyway, as mentioned I am located in Bavaria and I really can't believe the found IMSIs for Bermuda and Guam. Somehow strange ...

```
sqlite> select count(distinct imsi) as count, imsicountry from observations group by imsicountry order by count desc limit 25;
count       imsicountry
----------  -----------
62156       Germany    
1653        Bermuda    
1267        Poland     
733         Guam (Unite
548         Romania    
387         China      
306         Turkey     
289         France     
248         Spain      
231         Italy      
196         Austria    
165         Guernsey (U
152         Czech Repub
138         Bulgaria   
112         Netherlands
108         Hungary    
101         Russian Fed
83          Ukraine    
81          India      
80          Croatia    
76          Serbia     
76          Slovakia   
75          Latvia     
53          Greece     
51          Egypt   
```

E3V3A, 2018-10-25:
- There are a lot of americans in Germany, and they use US providers. 
 - The Wikipedia list is very poorly maintained and Bermuda in particular is wrong as it shows US and Jamaican MCC in that table. 
 - The web site: XXXX is also not perfect and I have complained before (in their github) about faulty data, but the maintainer refused to fix it. 

Bermuda from wiki:
```

MCC | MNC | Brand | Operator | Status | Bands (MHz) | References and notes
--- | --- | --- | --- | -- | -- | --
310 | 59 | Cellular One |   | Operational | CDMA | uses USA MCC
338 | 050 | Digicel Bermuda |   | Operational | GSM 1900 / UMTS 1900 / LTE 1900 | uses Jamaica MCC [72]
350 | 000 | CellOne | Bermuda Digital Communications Ltd. | Operational | GSM 1900 / UMTS 850 / LTE 850 | [38][5][6][72]

```

Solution? - Don't scrape from wikipedia! (And do some manual checking.)

sa7mon, 2018-11-13:
The most official authority for MCC's seems to be the ITU (International Telecommunications Union). They [publish bulletins regularly](http://www.itu.int/pub/T-SP-OB) and each bulletin seems to contain any changes to MCC/MNC's - just search the PDF for MNC. 

These bulletins don't contain the full current list, just the changes since the last bulletin. Maybe there's a way to scrape them all, I'm not sure. 

Interestingly, the Wikipedia page for MCC's does cite these bulletins frequently so it might not be that incorrect.

Oros42, 2018-11-13:
I don't use http://www.mcc-mnc.com/ because they didn't show update date.
So I don't know if it's up to date.

Interesting for http://www.itu.int/pub/T-SP-OB
They add follow us buttons but not RSS feed and JSON format to get data -_-
Not cool :-(

closed 2020-04-18 by Oros42