Oros42/IMSI-catcher #53: socket.error: [Errno 98] Address already in use

ericwang321, 2018-07-07:
hi     

 ubuntu 16.04 +  hackrf [Firmware Version: 2017.02.1 (API:1.02)]
------

1. 

  In  1st terminal  run 
  *sudo QT_X11_NO_MITSHM=1 grgsm_livemon --h*
-------

2. 

   In  2st terminal  run:
~/IMSI-catcher$ _**sudo python simple_IMSI-catcher.py**_ 
```
Nb IMSI ; TMSI-1 ; TMSI-2     ; IMSI   ; country  ; brand      ; operator ; MCC  ; MNC   ; LAC    ; CellId
Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 595, in <module>
    udpserver(port=options.port, prn=find_imsi)
  File "simple_IMSI-catcher.py", line 544, in udpserver
    sock.bind(server_address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use
```

3. 

 why tip " address already in use ",should I  change imsi-catcher.py  
 "udpdata, address = sock.recvfrom(4096)  " ?

thanks

ericwang321, 2018-07-07:
sudo python simple_IMSI-catcher.py -s 
---
  works
-------
   cannot find TSIM ,only imsi ,  why ?
----------
    how to find tsim relations to imsi ?
-------------

Oros42, 2018-07-08:
You can try this : 
```
sudo python simple_IMSI-catcher.py -s -a
```
```
$ python simple_IMSI-catcher.py -h
Usage: simple_IMSI-catcher.py: [options]

Options:
  -h, --help            show this help message and exit
  -a, --alltmsi         Show TMSI who haven't got IMSI (default  : false)
  -i IFACE, --iface=IFACE
                        Interface (default : lo)
  -m IMSI, --imsi=IMSI  IMSI to track (default : None, Example:
                        123456789101112 or "123 45 6789101112")
  -p PORT, --port=PORT  Port (default : 4729)
  -s, --sniff           sniff on interface instead of listening on port
                        (require root/suid access)
  -w SQLITE, --sqlite=SQLITE
                        Save observed IMSI values to specified SQLite file
```

ericwang321, 2018-07-18:
it works. simple_IMSI-catcher.py  can change port by -P paramter.
I learn the gsm basic theory,know TMSI with IMSI in packet. 
thanks.

closed 2018-07-18 by ericwang321