Oros42/IMSI-catcher #60: scan-and-livemon not working

SeqA000040, 2018-11-17:
Scan-and-livemon outputs "TypeError: an integer is required"
```
root@kali:~/IMSI-catcher# python scan-and-livemon
Locating potential GSM base station frequencies (this can take a few minutes).
Found 14 frequences
Traceback (most recent call last):
  File "scan-and-livemon", line 105, in <module>
    main()
  File "scan-and-livemon", line 85, in main
    freqs = select_freqs(list, numreceivers)
  File "scan-and-livemon", line 61, in select_freqs
    if 0 in sorted_list:
TypeError: an integer is required
```

afk11, 2018-11-19:
@SeqA000040 this worked for me (using python 2.7). my python isn't great, not even sure the message exception message appears :P

go to line 61, apply this 'diff':
```
-               if 0 in sorted_list:
-                       s = sorted_list[0]
+                print(sorted_list)
+                try:
+                       s = sorted_list[0]
                        freqs.append(s.freq)
                        operatorscount[(s.mcc, s.mnc)] = operatorscount[(s.mcc, s.mnc)] + 1
                        listcopy.remove(s)
-               count = count - 1
+                       count = count - 1
+                except Error:
+                       print "error occurred"
```

SeqA000040, 2018-11-20:
Thanks, that worked. Here's the updated file
[scan-and-livemon.zip](https://github.com/Oros42/IMSI-catcher/files/2601609/scan-and-livemon.zip)

Oros42, 2018-11-21:
Tell me if it's better with this fix.

loic773, 2020-09-23:
hello, I have a problem to make scan-and-livemon work I made the change that you explain above may put me File 
"scan-and-livemon", line 61
    if 0 in sorted_list:
                       ^
IndentationError: unindent does not match any outer indentation level"
could you tell me what i can do please
(i am in SMP Debian 5.8.7-1kali1 )

Oros42, 2020-09-24:
You didn't have to apply the previous fix because it's already done :
https://github.com/Oros42/IMSI-catcher/commit/b00a29ab4e68e54d7ea35ee5078c86985e440e87