Oros42/IMSI-catcher #63: Need to pass options to gsgsm_scanner

bferrell, 2018-12-12:
On my Raspberry PI 3B+ (Stretch) the option --args rtl must be passed to grgsm_scanner else it crashes.

I'm trying to see where that option can be set  for scan-and-livemon

Oros42, 2018-12-12:
I think you have to had it in options.args between line 32 and 33.
I don't like this part of code because we have conflict with argument_parser.
I haven't take the time to work on it :-(

bferrell, 2018-12-13:
Python isn't what I commonly code in so this may be a dumb question:

Starting at line 91, How about this:

             proc = subprocess.Popen(["grgsm_livemon_headless",
                 "--args rtl",
                 "--serverport=%d" % serverport,
                 "-f", str(freq)])
             procs.append(proc)
             serverport = serverport + 1

Possibly include a file, grgsmsettings.py with values?

Zarados, 2019-03-13:
@bferrell @Oros42 
Hi, I fix this problem.
I write "rtl" to the grgsm_scanner, let it only used rtl-sdr. 
And the other problem is grgsm_scanner usually not show any output.
If I run the  Kal (kalibrate-rtl) on the raspberry pi, it works.
Did you have any idea?

bferrell, 2019-04-07:
scan-and-livemon still crashes with fresh pull of gr-gsm