Oros42/IMSI-catcher #64: AttributeError: 'module' object has no attribute 'QWidget' error

kolet, 2018-12-24:
im on ubuntu 18.04 
installed all the dependenci 

> ImportError: No module named PyQt4

to solve this , i had to install python-qt4 ( sudo apt install python-qt4)

but when u do install it.
u get another error.

> Warning: failed to XInitThreads()
> Traceback (most recent call last):
>   File "/usr/bin/grgsm_livemon", line 45, in <module>
>     from gnuradio import qtgui
>   File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
>     from range import Range, RangeWidget
>   File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 67, in <module>
>     class RangeWidget(QtWidgets.QWidget):
> AttributeError: 'module' object has no attribute 'QWidget'

any idea

sa7mon, 2019-01-09:
Use Python3, not Python2

Oros42, 2019-01-12:
It's a problem with gr-gsm :
https://osmocom.org/projects/gr-gsm/wiki/Installation#Installation-from-packages-on-Debian-Testing-and-Ubuntu-1804
```
wget http://git.osmocom.org/gr-gsm/plain/apps/grgsm_livemon.grc
grcc -d . grgsm_livemon.grc
mv grgsm_livemon.py grgsm_livemon
```

closed 2020-04-18 by Oros42