Oros42/IMSI-catcher #67: Segmentation fault

jbdu94, 2019-02-25:
Hi Oros42

I am new to telephony and wanted to try your programs.
But, I have difficulties to run scan-and-livemon, gr-gsm and grgsm_livemon: I get segmentation fault each time.
I am using RTL2832 from Nooelec.

And when I run python simple_imsi_catcher.py, even after 20 minutes I get nothing.
Should I buy a bigger antenna? Any recommendation about antenna ? Or what am I doing wrong?

I have Kali linux, and I blacklisted the driver like described in Kubon forum.
rtl_test works fine.
Here is the procedure I followed for making my RTL working:

> Edit /etc/modprobe.d/raspi-blacklist.conf file and add the following lines
blacklist dvb_usb_rtl28xxu
blacklist dvb_usb_v2
blacklist rtl_2830
blacklist rtl_2832
blacklist r820t

sudo apt-get install git build-essential cmake libusb-1.0-0-dev
git clone git://git.osmocom.org/rtl-sdr.git
cd rtl-sdr
mkdir build
cd build
cmake ../ -DDETACH_KERNEL_DRIVER=ON -DINSTALL_UDEV_RULES=ON
make
sudo make install
sudo ldconfig

Here are outputs I have, in case it is useful for you, to help me solve the issue:

`grgsm_livemon
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
[INFO] Using format CF32.
Segmentation fault
`
`./scan-and-livemon 
Locating potential GSM base station frequencies (this can take a few minutes).
Segmentation fault

`grgsm_scanner

Segmentation fault
`

`python simple_IMSI-catcher.py --sniff
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC  ; MNC   ; LAC    ; CellId

`

I tried with and without sniff, but the same.
I hope that you can help me to solve this.

By the way, is there a way to obtain imei too? I would like to just scan for a device/number in particular (mine), would it be possible? Or should I query from results (when I will get them)?

Thanks a lot in advance for your help and really congrats for your program!

dhk175, 2019-04-12:
I'm having the same issue. I either get segmentation fault, or just a return of 0 stations found after about 15min. Tried installing on different os. Made sure that my USB is recognized too.

jbdu94, 2019-06-11:
Now I don't have segfault anymore, but simple_IMSI_Catcher.py still doesn't return anything. While grgsm_scanner works fine

What could be the reason why I don't get any result with the script? I am on linux, grgsm_scanner works fine but I would like to see if I find my IMSI among the IMSI returned by the script, however it doesn't show anything

dhk175, 2019-06-11:
Same issue with not seeing any output with it working as well.

Oros42, 2019-06-12:
I have you try :
```python simple_IMSI-catcher.py```
(no ```sudo``` and ```-s```)

jbdu94, 2019-06-12:
Hi Oros42 
thanks but without  -s, it still doesn't display anything, even after 20 min !
I am root by default on my distro
It did work once, then no more.
The dongle is fine, I can run grgsm_scanner 
I really want to use your program, I run kali linux (debian based)
By the way, is there a way in your program to display signal, like grgsm_scanner propose? It would be cool to get the IMSI and his signal strength from his cell, but first I would like to be able to obtain results from your program
My dongle is RTL2832

dhk175, 2019-06-13:
We are both clearly missing a step somewhere here. I have updated and
upgraded several times. Updated my rolling Kali repository. Tried running
the installation multiple times, but have the same issue that no data ever
starts showing up. Is it possible that we have installed from the wrong
list. I have tried it on Ubuntu as well, but same result. Any thing we need
to enable or disable? I have had trouble with blacklisting one of the
configurations with the RTL-SDR, can't remember the name right now, but no
luck. With a current version of Kali, which set of instructions do we
follow? I would like to just try it again with a new image.

Oros42, 2019-06-13:
@jbdu94 euuuh did you run ```grgsm_livemon``` ?
Because ```grgsm_livemon``` show you the signal strength and redirect cell's packets to your localhost.
And ```simple_IMSI-catcher.py``` only listen your localhost and not directly the cell.
You have to find the good frequency in ```grgsm_livemon```.

jbdu94, 2019-06-14:
@Oros42 
I ran grgsm_scanner, checked the best frequency (where the signal has a value closer to zero), and I put this frequency into grgsm_livemon.

Nothing however has been displayed in simple_imsi_catcher, even after 20min

NB: I changed a few time the freq in grgsm_livemon, the last one was the good one, it displayed stuff in hexadecimal, but no IMSI showed up in simple_imsi_catcher

#################################################################################
Here is the output of grgsm_livemon

grgsm_livemon
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:1,0), Device or resource busy.

Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
[R82XX] No valid PLL values for 1799425000 Hz!
r82xx_set_freq: failed=-1
[R82XX] PLL not locked!
[R82XX] No valid PLL values for 1991225000 Hz!
r82xx_set_freq: failed=-1
[R82XX] No valid PLL values for 1831425000 Hz!
r82xx_set_freq: failed=-1
[R82XX] No valid PLL values for 1783425000 Hz!
r82xx_set_freq: failed=-1
 49 06 1b 27 93 02 f8 41 03 f3 e8 02 3c 64 62 4d b9 00 00 85 00 8b 2b
 05 06 20 ab 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 05 06 20 ab 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 05 06 20 eb 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 00 eb 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 00 eb 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 00 eb 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 01 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b

Oros42, 2020-04-18:
```2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b``` == no data
so no IMSI to decode
Change the frequency

closed 2020-04-18 by Oros42