Oros42/IMSI-catcher #68: scapy.error.Scapy_Exception: BIOCSETIF failed on lo

nsklyarov, 2019-03-17:
trying run script on OS X High Sierra 10.13.6 (17G65)

```
Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 533, in <module>
    sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi_from_pkg, store=0)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/scapy/sendrecv.py", line 836, in sniff
    *arg, **karg)] = iface
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/scapy/arch/bpf/supersocket.py", line 72, in __init__
    raise Scapy_Exception("BIOCSETIF failed on %s" % self.iface)
scapy.error.Scapy_Exception: BIOCSETIF failed on lo
```

nsklyarov, 2019-05-02:
please answer me if my question is incomprehensible

Oros42, 2019-06-12:
Have you try ? :
```python scan-and-livemon```
(no sudo)

closed 2020-04-18 by Oros42