Oros42/IMSI-catcher #70: why not

root0day, 2019-03-31:
[image]
[image]
[image]
[image]

can you tell me how to solve this probiem?
not a conmputer majior
please

root0day, 2019-03-31:
root@ubuntu:/home/ubuntu# uname -a
Linux ubuntu 4.18.0-15-generic #16~18.04.1-Ubuntu SMP Thu Feb 7 14:06:04 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
root@ubuntu:/home/ubuntu#

Oros42, 2019-06-12:
For Ubuntu 18.04, remove ptrkrysik's repository and take a look to https://osmocom.org/projects/gr-gsm/wiki/Installation

closed 2020-04-18 by Oros42