Oros42/IMSI-catcher #72: Insert data into mysql

ciphercc, 2019-05-17:
Hi,

how to insert data into mysql

Oros42, 2019-06-12:
1- find documentation about python and mysql
2- add new code or rewrite part with sqlite

Toyosie, 2020-01-24:
import mysql.connector

conn = mysql.connector.connect(
    host ="localhost",
    user = "root",
    password = "",
    db = "test"
)

e = conn.cursor()

i_d = int(input("Id: ")
celebrity = str(input("Celebrity: "))
job = input ("Job: ")

e.execute ("""INSERT INTO model (ID, Celebrity, job) Values ({}, {}, {})""".format(i_d, celebrity, job))

conn.commit()
conn.close()
print("Weldone! inserted successfully")

my error:
ProgrammingError                          Traceback (most recent call last)
<ipython-input-5-fbc5a3e266be> in <module>
     16 
     17 idcheck = "SELECT ID, Celebrity, job FROM model WHERE Celebrity = %s" % celebrity
---> 18 e.execute(idcheck)
     19 
     20 conn.commit()

~\Anaconda3\lib\site-packages\mysql\connector\cursor.py in execute(self, operation, params, multi)
    567 
    568         try:
--> 569             self._handle_result(self._connection.cmd_query(stmt))
    570         except errors.InterfaceError:
    571             if self._connection._have_next_result:  # pylint: disable=W0212

~\Anaconda3\lib\site-packages\mysql\connector\connection.py in cmd_query(self, query, raw, buffered, raw_as_string)
    588         if not isinstance(query, bytes):
    589             query = query.encode('utf-8')
--> 590         result = self._handle_result(self._send_cmd(ServerCmd.QUERY, query))
    591 
    592         if self._have_next_result:

~\Anaconda3\lib\site-packages\mysql\connector\connection.py in _handle_result(self, packet)
    476             return self._handle_eof(packet)
    477         elif packet[4] == 255:
--> 478             raise errors.get_exception(packet)
    479 
    480         # We have a text result set

closed 2020-04-18 by Oros42