Oros42/IMSI-catcher #74: gnuradio 3.8 compatibility

lipenasc, 2019-10-19:
Recently gnuradio was updated to release 3.8.
Using @velichkov tips in [https://github.com/ptrkrysik/gr-gsm/issues/480] I think I fixed the troubles with gnuradio and gr-grsm packages compatibility, but when I try to run scan-and-livemon...

> root@kali:~/SDR/IMSI-catcher# python scan-and-livemon
> Locating potential GSM base station frequencies (this can take a few minutes).
> Traceback (most recent call last):
>   File "scan-and-livemon", line 102, in <module>
>     main()
>   File "scan-and-livemon", line 78, in main
>     list = find_gsm_bases()
>   File "scan-and-livemon", line 30, in find_gsm_bases
>     scanner = imp.load_source('scanner', grgsm_scanner_path)
>   File "/usr/local/bin/grgsm_scanner", line 24, in <module>
>     from gnuradio import blocks
> ImportError: No module named gnuradio

Using Kali in VMWare.
I think it's a matter of compatibility

innerfirebg, 2019-12-09:
i have the some problem... :(

innerfirebg, 2019-12-09:
I have the same problem... :(

MrNonoss, 2019-12-30:
Any news about it ?

lipenasc, 2020-01-03:
Sorry, I've been a little busy in other projects...
Any news I have I'll let you know

alphafox02, 2020-07-05:
Most likely you have just python3 / GNU Radio 3.8 installed. I have the same issue with this combination. Instead of python I type python3 before scan and livemon, but no idea if it actually works. I think maybe the IMSI catcher could use some updating or checking against the gnuradio 3.8 gr-gsm branch mentioned above.

alphafox02, 2020-07-05:
```
### With version of gr-gsm >= 0.41.2-1

Open 2 terminals.
In terminal 1

python3 simple_IMSI-catcher.py

You can add -h to display options.

In terminal 2

python scan-and-livemon
```
Even if you replace the last line with python3 and try to run it you end up with the error that the port is already in use. So as far as I can tell the gr-gsm 3.8 branch works on it's own, but not in combination with the IMSI catcher script. 

This method with python3 / GNU Radio 3.8 and the 3.8 gr-gsm branch runs fine, but I've not seen any sign that the IMSI catcher script is actually doing anything,

With an old version of gr-gsm

```
Open 2 terminals.
In terminal 1

sudo python3 simple_IMSI-catcher.py --sniff

You can add -h to display options.

In terminal 2, search a frequency to listen :

grgsm_scanner
```

alphafox02, 2020-07-05:
I’ve tried every which way I can think of using it, it appears the IMSI script is not compatible with the Gnuradio 3.8 grgsm branch referenced above. Grgsm on its own runs fine but either option available to use the IMSI script results in either the “port in use” or when trying option 1 this happens 

Nb IMSI ; TMSI-1 ; TMSI-2 ; IMSI ; country ; brand ; operator ; MCC ; MNC ; LAC ; CellId ; Timestamp
Traceback (most recent call last):
File "simple_IMSI-catcher.py", line 538, in <module>
sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi_from_pkt, store=0)
File "/usr/lib/python3/dist-packages/scapy/sendrecv.py", line 972, in sniff
sniffer._run(*args, **kwargs)
File "/usr/lib/python3/dist-packages/scapy/sendrecv.py", line 925, in _run
session.on_packet_received(p)
File "/usr/lib/python3/dist-packages/scapy/sessions.py", line 47, in on_packet_received
result = self.prn(pkt)
File "simple_IMSI-catcher.py", line 498, in find_imsi_from_pkt
find_imsi(udpdata)
File "simple_IMSI-catcher.py", line 429, in find_imsi
t.register_imsi(gsm.arfcn, imsi1, imsi2, tmsi1, tmsi2, p)
File "simple_IMSI-catcher.py", line 225, in register_imsi
self.pfields(str(n), self.str_tmsi(tmsi1), self.str_tmsi(tmsi2), imsi1, str(self.mcc), str(self.mnc), str(self.lac), str(self.cell), p)
File "simple_IMSI-catcher.py", line 162, in pfields
self.output_function(cpt, tmsi1, tmsi2, imsi, imsicountry, imsibrand, imsioperator, mcc, mnc, lac, cell, now, packet)
File "simple_IMSI-catcher.py", line 151, in output
print("{:7s} ; {:10s} ; {:10s} ; {:17s} ; {:12s} ; {:10s} ; {:21s} ; {:4s} ; {:5s} ; {:6s} ; {:6s} ; {:s}".format(str(cpt), tmsi1, tmsi2, imsi, imsicountry.encode('utf-8'), imsibrand.encode('utf-8'), imsioperator.encode('utf-8'), str(mcc), str(mnc), str(lac), str(cell), now.isoformat()))
TypeError: unsupported format string passed to bytes.__format__

Oros42, 2020-09-08:
I have found how to run python3 + GNU Radio 3.8 + simple_IMSI-catcher.py on LinuxMint 20 / Ubuntu 20.04 \o/
See this https://github.com/Oros42/IMSI-catcher#gr-gsm-for-ubuntu-2004
To setup gr-gsm :
```
sudo apt-get install -y \
    cmake \
    autoconf \
    libtool \
    pkg-config \
    build-essential \
    python-docutils \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    gnuradio-dev \
    gr-osmosdr \
    libosmocore-dev \
    liborc-0.4-dev

git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..
make -j 4
sudo make install
sudo ldconfig
echo 'export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH' >> ~/.bashrc
```

microwaven, 2021-08-15:
The README.md files states that for gnuradio 3.7, we shall use https://git.osmocom.org/gr-gsm but that requires Gnuradio 3.8 since May.

Both https://github.com/velichkov/gr-gsm.git and https://git.osmocom.org/gr-gsm/ seem to require Gnuradio 3.8. Interestingly, Vasil has spent an effort into https://git.osmocom.org/gr-gsm/ recently but not into https://github.com/velichkov/gr-gsm . Which one to use?

Somewhat difficult to build for Debian Buster these days that uses 3.7.

Oros42, 2021-08-18:
aaaah...
I'm not the author of gr-gsm. So I don't know :-/
Have you ty :
```
sudo apt install gr-gsm
```
or
```
sudo add-apt-repository -y ppa:ptrkrysik/gr-gsm
sudo apt update
sudo apt install gr-gsm
```

PARAMJAI, 2023-02-17:
> I have found how to run python3 + GNU Radio 3.8 + simple_IMSI-catcher.py on LinuxMint 20 / Ubuntu 20.04 \o/ See this https://github.com/Oros42/IMSI-catcher#gr-gsm-for-ubuntu-2004 To setup gr-gsm : [...]

PARAMJAI, 2023-02-17:
I am having this error message on this "-- Configuring incomplete, errors occurred!"
[image]

InfinityJammyyy, 2024-10-15:
Instead of compiling gr-gsm from Github there are packages on Debian (https://packages.debian.org/bookworm/gr-gsm) and on Arch Linux through the AUR (https://aur.archlinux.org/packages/gr-gsm-git) that seem to work.

The compilation error seems to be an incompatibility thing between gr-gsm and the newer versions of gnuradio

PS I have only tested this "fix" on Manjaro and it seems to work just fine, I have not tested this fix on Debian.

alphafox02, 2024-10-15:
If you’re trying to compile gr-gsm for the latest gnuradio 3.10.. use the PR labeled 3.10-3.11 support to do so. The PR still not been pulled into main, so if you’re using main and trying to compile on recent distributions it’s almost guaranteed to fail.