Oros42/IMSI-catcher #75: Only wrong IMSI numbers

DavidBain1234, 2019-10-20:
Hi there . I am using a hackrf grgsm livemon and simple imsi catcher.  I am in a location that only has one cell tower only GSM. I have 5 phones.  I can see 5 imsi numbers coming up but none of them are correct they are not the same numbers as collected from the sim cards. If I look at the app on the phone it gives me the imsi of the sim.  I have also spoken to a friend of mine that works for the local provider and he confirmed my imsi number as the same as what the app gives me. The only think that is correct is the cell tower information but not the imsi. The first two section of the number are correct country and provider but the rest of the number seems to be completely out not even close to the number on the phone.

flanneryb, 2019-10-20:
David,

Without knowing what your program looks like. Are you sure that you’re not
pulling the IMEI?   In order to pull the IMSI you would need to be
interfacing with the devices and I’m not sure that you want to do that.

Brendan

DavidBain1234, 2019-10-20:
> David, Without knowing what your program looks like. Are you sure that you’re not pulling the IMEI? In order to pull the IMSI you would need to be interfacing with the devices and I’m not sure that you want to do that. Brendan
> -- V/R Brendan

no its the IMSI number I'm getting I know what the emei number is and that belongs to the cell not the sim the IMSI is the number on the sim card. its not difficult to get with the right app.

Oros42, 2019-10-20:
To help us to fix it, run wireshark and capture the network communication.
Find your IMSI in packets. And tell us what the packet look like.
Examples of packets in simple_IMSI-catcher.py line 384 to 475.

DavidBain1234, 2019-10-21:
Hi Oros42.   Found the problem it was on my side.  I have a question for you though. I need to add two three more fields to simple imsi catcher.py I've tried to add them it's prints out when I start but as soon as Th ere is data added I get the message can't change from auto to manual insurt.  I need to add a field ID and a device ID the field ID is auto increment number based on the last number in the Sqlite db the device ID is static field insured into the py script on install it's the Sr number of the hackrf been used.   Would you be able help me add those fields .

Oros42, 2019-10-21:
Unfortunately, I don't have enough time to do help you :-(
Have you change line 142 ?
```
self.sqlcon.execute("CREATE TABLE...
```
Have you remove the sqlite DB ?

If you publish your code, perhaps someone else will help you :-)

DavidBain1234, 2019-10-21:
No problem thanks anyway I'll post the code and see if anyone can help me with.

DavidBain1234, 2019-10-21:
Below is the code all I need to do is add two fields an auto increment number this will be the row ID  at the end and a static number that will be inserted into each entry this will be the row device ID. I have tried to add these two fields but each time I do. I get the error cant change from auto to manual entry.  

#!/usr/bin/python
# -*- coding: utf-8 -*-
# Author: Oros
# Contributors : puyoulu, 1kali2kali, petterreinholdtsen
# 2018/08/18
# License : CC0 1.0 Universal

"""
This program shows you IMSI numbers of cellphones around you.

/!\ This program was made to understand how GSM network work. Not for bad hacking !
"""

import ctypes
import json
from optparse import OptionParser
import datetime
import io
import socket

imsitracker = None

class tracker:
	imsistate = {}
	# phones
	imsis=[] # [IMSI,...]
	tmsis={} # {TMSI:IMSI,...}
	nb_IMSI=0 # count the number of IMSI

	mcc=""
	mnc=""
	lac=""
	cell=""
	country=""
	brand=""
	operator=""

	show_all_tmsi = False
	mcc_codes = None
	sqlcon = None

	ouput_function=None

	def __init__(self):
		self.load_mcc_codes()
		self.track_this_imsi("")
		self.ouput_function=self.ouput

	def set_ouput_function(self, new_ouput_function):
		# New ouput function need this field :
		# cpt, tmsi1, tmsi2, imsi, imsicountry, imsibrand, imsioperator, mcc, mnc, lac, cell, packet=None
		self.ouput_function=new_ouput_function

	def track_this_imsi(self, imsi_to_track):
		self.imsi_to_track = imsi_to_track
		self.imsi_to_track_len=len(imsi_to_track)

	# return something like '0xd9605460'
	def str_tmsi(self, tmsi):
		if tmsi != "":
			new_tmsi="0x"
			for a in tmsi:
				c=hex(ord(a))
				if len(c)==4:
					new_tmsi+=str(c[2])+str(c[3])
				else:
					new_tmsi+="0"+str(c[2])
			return new_tmsi
[... 441 lines cut ...]
		imsi="9"+options.imsi.replace(" ", "")
		imsi_to_track_len=len(imsi)
		if imsi_to_track_len%2 == 0 and imsi_to_track_len > 0 and imsi_to_track_len <17:
			for i in range(0, imsi_to_track_len-1, 2):
				imsi_to_track+=chr(int(imsi[i+1])*16+int(imsi[i]))
			imsi_to_track_len=len(imsi_to_track)
		else:
			print("Wrong size for the IMSI to track!")
			print("Valid sizes :")
			print("123456789101112")
			print("1234567891011")
			print("12345678910")
			print("123456789")
			print("1234567")
			print("12345")
			print("123")
			exit(1)
	imsitracker.track_this_imsi(imsi_to_track)
	if options.sniff:
		from scapy.all import sniff, UDP
		imsitracker.header()
		sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi_from_pkg, store=0)
	else:
		imsitracker.header()
		udpserver(port=options.port, prn=find_imsi)

Mihail354, 2020-03-01:
[

closed 2020-04-18 by Oros42