Oros42/IMSI-catcher #77: Cell Location

dieskaaphond, 2020-01-06:
Good day Sir,

I am a farmer in South Africa and not a programmer or coder.
I am trying to build a device to locate intruders accessing my farm and committing theft, murder etc.

I have made the IMSI catcher work. How can I get it now to give directional or physical location of mobile device using IMSI catcher.

Can you assist?

Oros42, 2020-01-06:
My project listen cell tower. So when you get an ISMI, you only know that a mobile device is around one specific cell tower.
But you didn't know exactly where the mobile device is.
I think for you, a passive radar would be better suited.
Something like that :
https://www.rtl-sdr.com/measuring-traffic-in-a-neighborhood-with-kerberossdr-and-passive-radar/

Locate mobile device is something that I want to do. But in 5 years, I don't found the time do to this :'-(

Keep me informed of your progress ;-)

dieskaaphond, 2020-01-08:
Thank you, will do. Appreciate your assistance

dieskaaphond, 2020-01-12:
Hi,

Thank you for your reply but this will not work as they move on foot and not in vehicles. There is nothing to bounce the signal off.

Kind regards

Andre Swanepoel

Oros42, 2020-01-14:
You can track lot of more than vehicles. https://othernet.is/products/kerberossdr-4x-coherent-rtl-sdr
You can locate the source of a signal (ex phone for 2G/3G).

zaidragie, 2020-01-27:
Good day
I'm also looking to figure out a way to track a mobile device. 
Would you be able to point me in the correct direction on doing this

closed 2020-04-18 by Oros42