Oros42/IMSI-catcher #79: IMSI run but cant read data from gr gsm

asyooty249, 2020-01-09:
[image]
dears 
need your help why IMSi cant read the data from gr gsm as you can see in the picture the ge gsm have data but i cant see it in IMSI
any advice to solve the issue

asyooty249, 2020-01-09:
dears 

thanks for your support i had fix the issue but now is no showing the phone number in the IMSI .

asyooty249, 2020-01-09:
[image]
same issue

tombaki, 2020-02-05:
@asyooty249 
i'v same issue but i canno't get any data from console what did u do to fix that
root@kali:~/imsi/imsi# ./simple_IMSI-catcher.py --sniff
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC  ; MNC   ; LAC    ; CellId ; Timestamp

Oros42, 2020-04-18:
No data == no IMSI send by the cell tower.
Try to change the frequency

Cell Id take few seconds to update.

closed 2020-04-18 by Oros42