Oros42/IMSI-catcher #109: scan and livemon

loic773, 2020-09-23:
I have a problem running scan-and-livemon here is the result could you tell me what to do
File "scan-and-livemon", line 102, in <module>
    main()
  File "scan-and-livemon", line 78, in main
    list = find_gsm_bases()
  File "scan-and-livemon", line 30, in find_gsm_bases
    scanner = imp.load_source('scanner', grgsm_scanner_path)
  File "/usr/local/bin/grgsm_scanner", line 24, in <module>
    from gnuradio import blocks
ImportError: No module named gnuradio

i tried to fix this by changing the python script as explained in issue # 8 ?may nothing helps

Oros42, 2020-09-24:
It's a gr-gsm error.
Try this setup : https://github.com/Oros42/IMSI-catcher#gr-gsm-for-ubuntu-2004

loic773, 2020-09-27:
thanks for your reply, scan and livemon  now works, however it cannot find any frequency did you have any idea what to do?

pranjal-b, 2021-06-18:
@loic773 I'm facing the same issue.