Oros42/IMSI-catcher #110: grgsm_livemon errore help me

cesare3338, 2020-11-02:
root@kali:~# grgsm_scanner

root@kali:~# sudo python3 simple_IMSI-catcher.py --sniff
python3: can't open file 'simple_IMSI-catcher.py': [Errno 2] No such file or directory
root@kali:~# grgsm_livemon -f <your_frequency>M
-bash: your_frequency: File o directory non esistente
root@kali:~# grgsm_livemon -f 938.2M
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-3
[ERROR] avahi_service_browser_new() failed: Bad state
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 385, in <module>
    main()
  File "/usr/bin/grgsm_livemon", line 361, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/bin/grgsm_livemon", line 113, in __init__
    args="numchan=" + str(1) + " " + str(grgsm.device.get_default_args(args))
  File "/usr/lib/python3/dist-packages/grgsm/device.py", line 55, in get_default_args
    raise RuntimeError("Unable to find any supported SDR devices")
RuntimeError: Unable to find any supported SDR devices
root@kali:~# sudo python simple_IMSI-catcher.py -s
python: can't open file 'simple_IMSI-catcher.py': [Errno 2] No such file or directory

result grgsm_livemon does not open

Oros42, 2020-11-02:
I recommend you to take more time to read what you post. And to read error messages.
I think your are not in the good directory.
And when you are root, no need of `sudo`.

cesare3338, 2020-11-03:
novice user, I can't get it to open grgsm_livemon

irisroques, 2020-11-15:
Hey i'm using the docker image and also can't run grgsm_livemoon. I get a comand not found message.

Oros42, 2020-11-15:
@irisroques did you do this :
```bash
docker pull atomicpowerman/imsi-catcher
docker run -ti --net=host -e DISPLAY=$DISPLAY --privileged -v /dev/bus/usb:/dev/bus/usb  atomicpowerman/imsi-catcher bash
grgsm_livemoon
```
?

Warhammar, 2021-10-06:
O] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[ERROR] avahi_service_browser_new() failed: Bad state
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 385, in <module>
    main()
  File "/usr/bin/grgsm_livemon", line 361, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/bin/grgsm_livemon", line 113, in __init__
    args="numchan=" + str(1) + " " + str(grgsm.device.get_default_args(args))
  File "/usr/lib/python3/dist-packages/grgsm/device.py", line 55, in get_default_args
    raise RuntimeError("Unable to find any supported SDR devices")
RuntimeError: Unable to find any supported SDR devices

I Have a USB dongle exactly like this one: [https://www.youtube.com/watch?v=UjwgNd_as30](url)
What do i not see?

Warhammar, 2021-11-08:
> @irisroques did you do this :
> 
> ```shell
> docker pull atomicpowerman/imsi-catcher
> docker run -ti --net=host -e DISPLAY=$DISPLAY --privileged -v /dev/bus/usb:/dev/bus/usb  atomicpowerman/imsi-catcher bash
> grgsm_livemoon
> ```
> 
> ?

I did.. and comes.

`QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 365, in <module>
    main()
  File "/usr/local/bin/grgsm_livemon", line 353, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/local/bin/grgsm_livemon", line 107, in __init__
    self.rtlsdr_source_0 = osmosdr.source( args="numchan=" + str(1) + " " + str(grgsm.device.get_default_args(args)) )
  File "/usr/local/lib/python2.7/dist-packages/grgsm/device.py", line 55, in get_default_args
    raise RuntimeError("Unable to find any supported SDR devices")
RuntimeError: Unable to find any supported SDR devices
root@kali:/# 
`