Oros42/IMSI-catcher #112: python import error

killerdevildog11, 2020-11-20:
running grgsm_scanner gives the following error:
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 33, in <module>
    import grgsm
ModuleNotFoundError: No module named 'grgsm'

Oros42, 2020-11-21:
How did you setup grgsm ?
How did you run grgsm_scanner ?

killerdevildog11, 2020-11-21:
https://github.com/Oros42/IMSI-catcher
I followed every instruction given on the front page
grgsm_scanner is on my system but i think its not finding the grgsm python package to import to that file. /usr/local/grgrsm_scanner

Oros42, 2020-11-22:
There are 4 ways to setup gr-gsm.
So, if you want my help, please answer my previous questions

Warhammar, 2021-11-05:
I did it over Docker!
So is there a simple way to do this?

Warhammar, 2021-11-05:
grgsm_livemon                                       
terminate called after throwing an instance of 'std::runtime_error'
  what():  rpcmanager: Aggregator not in use, and a rpc booter is already registered

zsh: abort      grgsm_livemon