Oros42/IMSI-catcher #121: Couldn't make scan and livemon work

DJOCKER-FACE, 2021-01-17:
└─$ python3 scan-and-livemon  
/home/ThakaliPC/Desktop/IMSI/IMSI-catcher/scan-and-livemon:18: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Locating potential GSM base station frequencies (this can take a few minutes).
zsh: abort      python3 scan-and-livemon

DJOCKER-FACE, 2021-01-17:
And when i have gr-gsm running and try to start simple imsi catcher.py i get this:

python3 simple_IMSI-catcher.py                                                                      134 ⨯

Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC  ; MNC   ; LAC    ; CellId ; Timestamp
Traceback (most recent call last):
  File "/home/ThekaliPC/Desktop/IMSI/IMSI-catcher/simple_IMSI-catcher.py", line 612, in <module>
    udpserver(port=options.port, prn=find_imsi)
  File "/home/ThekaliPC/Desktop/IMSI/IMSI-catcher/simple_IMSI-catcher.py", line 551, in udpserver
    sock.bind(server_address)
OSError: [Errno 98] Address already in use

DJOCKER-FACE, 2021-01-17:
What i understood is when gr-gsm running/using the rtl the Simple IMSI can't open and when i start IMSI catcher first gr-gsm dosen't work, Finally when trying scan-and-livemon dosen't open at all. Any solution ?

Oros42, 2021-01-19:
Use this :
```
sudo python3 simple_IMSI-catcher.py -s
```
and
```
grgsm_livemon
```

closed 2021-01-19 by Oros42

DJOCKER-FACE, 2021-01-24:
This 
`sudo python3 simple_IMSI-catcher.py -s`

Gives that:

──(acer㉿kali)-[~/Desktop/IMSI/IMSI-catcher]
└─$ sudo python3 simple_IMSI-catcher.py -s                                                        130 ⨯

Traceback (most recent call last):
  File "/home/acer/Desktop/IMSI/IMSI-catcher/simple_IMSI-catcher.py", line 607, in <module>
    from scapy.all import sniff, UDP
  File "/usr/lib/python3/dist-packages/scapy/all.py", line 16, in <module>
    from scapy.arch import *
  File "/usr/lib/python3/dist-packages/scapy/arch/__init__.py", line 27, in <module>
    from scapy.arch.bpf.core import get_if_raw_addr
  File "/usr/lib/python3/dist-packages/scapy/arch/bpf/core.py", line 30, in <module>
    LIBC = cdll.LoadLibrary(find_library("libc"))
  File "/usr/lib/python3.9/ctypes/util.py", line 341, in find_library
    _get_soname(_findLib_gcc(name)) or _get_soname(_findLib_ld(name))
  File "/usr/lib/python3.9/ctypes/util.py", line 147, in _findLib_gcc
    if not _is_elf(file):
  File "/usr/lib/python3.9/ctypes/util.py", line 99, in _is_elf
    with open(filename, 'br') as thefile:
FileNotFoundError: [Errno 2] No such file or directory: b'liblibc.a'

I've done my small research apparently it's a scapy bug but i couldn't find how to fix it

Oros42, 2021-01-25:
Why did you add `130 ⨯`? It's not a valid option.

Try this
```
sudo python3 simple_IMSI-catcher.py -s -iface lo
```

DJOCKER-FACE, 2021-02-05:
`$ sudo python3 simple_IMSI-catcher.py -s -iface lo   `
Traceback (most recent call last):
  File "/home/acer/Desktop/IMSI/IMSI-catcher/simple_IMSI-catcher.py", line 607, in <module>
    from scapy.all import sniff, UDP
  File "/usr/lib/python3/dist-packages/scapy/all.py", line 16, in <module>
    from scapy.arch import *
  File "/usr/lib/python3/dist-packages/scapy/arch/__init__.py", line 27, in <module>
    from scapy.arch.bpf.core import get_if_raw_addr
  File "/usr/lib/python3/dist-packages/scapy/arch/bpf/core.py", line 30, in <module>
    LIBC = cdll.LoadLibrary(find_library("libc"))
  File "/usr/lib/python3.9/ctypes/util.py", line 341, in find_library
    _get_soname(_findLib_gcc(name)) or _get_soname(_findLib_ld(name))
  File "/usr/lib/python3.9/ctypes/util.py", line 147, in _findLib_gcc
    if not _is_elf(file):
  File "/usr/lib/python3.9/ctypes/util.py", line 99, in _is_elf
    with open(filename, 'br') as thefile:
FileNotFoundError: [Errno 2] No such file or directory: b'liblibc.a'

Still nothing what i'm i doing wrong ?

Oros42, 2021-02-08:
I suggest you to use an other version of python.
https://bbs.archlinux.org/viewtopic.php?id=261816