Oros42/IMSI-catcher #122: scapy error

nukil33t, 2021-03-18:
Try to start with `python3 simple_IMSI-catcher.py --sniff`

> Traceback (most recent call last):
  File "/home/nukikal/IMSI-catcher/simple_IMSI-catcher.py", line 607, in <module>
    from scapy.all import sniff, UDP
  File "/usr/local/lib/python3.9/dist-packages/scapy/all.py", line 16, in <module>
    from scapy.arch import *
  File "/usr/local/lib/python3.9/dist-packages/scapy/arch/__init__.py", line 27, in <module>
    from scapy.arch.bpf.core import get_if_raw_addr
  File "/usr/local/lib/python3.9/dist-packages/scapy/arch/bpf/core.py", line 30, in <module>
    LIBC = cdll.LoadLibrary(find_library("libc"))
  File "/usr/lib/python3.9/ctypes/util.py", line 341, in find_library
    _get_soname(_findLib_gcc(name)) or _get_soname(_findLib_ld(name))
  File "/usr/lib/python3.9/ctypes/util.py", line 147, in _findLib_gcc
    if not _is_elf(file):
  File "/usr/lib/python3.9/ctypes/util.py", line 99, in _is_elf
    with open(filename, 'br') as thefile:
FileNotFoundError: [Errno 2] No such file or directory: b'liblibc.a'
> 
grgsm_livemon is running, scapy and scapy3 are installed

Oros42, 2021-03-18:
Same as #121
I suggest you to use an other version of python.
https://bbs.archlinux.org/viewtopic.php?id=261816

closed 2021-03-18 by Oros42

blshkv, 2022-02-07:
your suggestion is wrong. 
``` cdll.LoadLibrary(find_library("libc"))``` used to work in < python 3.9 by a mistake.
The correct code is to call with "lib" prefix:
```ctypes.cdll.LoadLibrary(find_library("c"))```

For more details, see the following https://bugs.python.org/issue42580

the bug is also fixed in the scapy-2.4.5 for sure.

Please remove your "python < 3.9" requirement in the main page.

blshkv, 2022-02-08:
I haven't, but i tried scapy with python 3.10. it works. So catcher should work too