Oros42/IMSI-catcher #123: Address already in use

greengit-05, 2021-03-20:
When I run in 2-nd terminal, get error:

```
kali@kali:~/IMSI-catcher-master$ python simple_IMSI-catcher.py                                 
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC  ; MNC   ; LAC    ; CellId
Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 536, in <module>
    udpserver(port=options.port, prn=find_imsi)
  File "simple_IMSI-catcher.py", line 485, in udpserver
    sock.bind(server_address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use

```
I read [here](https://github.com/ptrkrysik/gr-gsm/issues/442) but grgsm_livemon is must be run, I can't kill them.
Also I read [here](https://github.com/Oros42/IMSI-catcher/issues/53) but it's don't help me.
Help me please to solve this problem.

Oros42, 2021-03-21:
What did you have with :
```
sudo python3 simple_IMSI-catcher.py --sniff
```
?

greengit-05, 2021-03-21:
Thanks! I'm enter the key --sniff under root permission and it's working OK!

closed 2021-03-22 by Oros42