Oros42/IMSI-catcher #124: headless

invalidi11, 2021-04-14:
I am trying to accomplish the following:

python3 scan-and-livemon (on boot up and headless)

python3 simple_IMSI-catcher.py -t filename.txt (on boot up and headless)

Is there a way to accomplish this without a screen?

Oros42, 2021-04-14:
```
without a screen
```
You mean no screen print by simple_IMSI-catcher.py ?

invalidi11, 2021-04-15:
Yes no screen.

invalidi11, 2021-04-15:
I want scan-and-livemon as well as simple_IMSI-catcher.py to work @reboot with no screen.

Oros42, 2021-04-18:
scan-and-livemon is not really good because it catch some empty frequency.  
While waiting for I add a quite mode, you can do :
```bash
sudo python3 simple_IMSI-catcher.py -s > /dev/null
grgsm_livemon -f 925.4M
```
Set your own frequency.