Oros42/IMSI-catcher #130: sudo python3 ./simple_IMSI-catcher.py --sniff

NuranFerzeli, 2021-07-13:
hi my python3 version is 3.9.2.
 when i run this script (sudo python3 ./simple_IMSI-catcher.py --sniff ) i get this exception:
**Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC  ; MNC   ; LAC    ; CellId ; Timestamp
Traceback (most recent call last):
  File "/root/IMSI-catcher/./simple_IMSI-catcher.py", line 610, in <module>
    sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi_from_pkt, store=0)
  File "/usr/local/lib/python3.9/dist-packages/scapy/sendrecv.py", line 1263, in sniff
    sniffer._run(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/scapy/sendrecv.py", line 1210, in _run
    session.on_packet_received(p)
  File "/usr/local/lib/python3.9/dist-packages/scapy/sessions.py", line 108, in on_packet_received
    result = self.prn(pkt)
  File "/root/IMSI-catcher/./simple_IMSI-catcher.py", line 561, in find_imsi_from_pkt
    find_imsi(udpdata)
  File "/root/IMSI-catcher/./simple_IMSI-catcher.py", line 490, in find_imsi
    t.register_imsi(gsm.arfcn, imsi1, imsi2, tmsi1, tmsi2, p)
  File "/root/IMSI-catcher/./simple_IMSI-catcher.py", line 276, in register_imsi
    self.pfields(str(n), tmsi1, tmsi2, imsi1, str(self.mcc), str(self.mnc), str(self.lac), str(self.cell), p)
  File "/root/IMSI-catcher/./simple_IMSI-catcher.py", line 193, in pfields
    self.output_function(cpt, tmsi1, tmsi2, imsi, imsicountry, imsibrand, imsioperator, mcc, mnc, lac, cell, now, packet)
  File "/root/IMSI-catcher/./simple_IMSI-catcher.py", line 182, in output
    print("{:7s} ; {:10s} ; {:10s} ; {:17s} ; {:12s} ; {:10s} ; {:21s} ; {:4s} ; {:5s} ; {:6s} ; {:6s} ; {:s}".format(str(cpt), tmsi1, tmsi2, imsi, imsicountry.encode('utf-8'), imsibrand.encode('utf-8'), imsioperator.encode('utf-8'), str(mcc), str(mnc), str(lac), str(cell), now.isoformat()))
TypeError: unsupported format string passed to bytes.__format__
**

Oros42, 2021-07-13:
@lapolis : could you take a look at this ? (https://github.com/Oros42/IMSI-catcher/pull/128)

lapolis, 2021-07-13:
> hi my python3 version is 3.9.2.
> ... ...
> TypeError: unsupported format string passed to bytes.**format**
> **
Sorry about that, it should work on python2.7 which is apparently what I used while testing. So you might be able to use it for now. Later today i will fix it for python3.

NuranFerzeli, 2021-07-13:
@lapolis  hi) first i get imsi but after update kali linux version from 2020.3 to 2021.2 i got this exception( in advance " i run python simple_IMSI-catcher.py -s " and i can catch imsi

NuranFerzeli, 2021-07-13:
Hi dear what is your advice . I change my python version to 2.7?

lapolis, 2021-07-13:
> Hi dear what is your advice . I change my python version to 2.7?
> […](#)

Just for now but I will patch it later today cause using Python2.7 in 2021 is not something I want to live with

nocturneop15, 2021-07-13:
huh, just spent some time trying t figure out what did I wrong - glad it's known issue. wating for fix ,-) 

Thanks for great work on this project

NuranFerzeli, 2021-07-13:
i change my python version to 2.7.3 and i get this exception 
Traceback (most recent call last):
  File "simple_IMSI-catcher.py", line 608, in <module>
    from scapy.all import sniff, UDP
ImportError: No module named scapy.all

lapolis, 2021-07-13:
> i change my python version to 2.7.3 and i get this exception
> Traceback (most recent call last):
> File "simple_IMSI-catcher.py", line 608, in
> from scapy.all import sniff, UDP
> ImportError: No module named scapy.all

You need to install scapy on python2

NuranFerzeli, 2021-07-13:
i installed scapy on python3 .
but when i run script i get this error
root@kali:~/IMSI-catcher# python3 simple_IMSI-catcher.py -s
Nb IMSI ; TMSI-1     ; TMSI-2     ; IMSI              ; country      ; brand      ; operator              ; MCC  ; MNC   ; LAC    ; CellId ; Timestamp
Traceback (most recent call last):
  File "/root/IMSI-catcher/simple_IMSI-catcher.py", line 610, in <module>
    sniff(iface=options.iface, filter="port {} and not icmp and udp".format(options.port), prn=find_imsi_from_pkt, store=0)
  File "/usr/local/lib/python3.9/dist-packages/scapy/sendrecv.py", line 1263, in sniff
    sniffer._run(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/scapy/sendrecv.py", line 1210, in _run
    session.on_packet_received(p)
  File "/usr/local/lib/python3.9/dist-packages/scapy/sessions.py", line 108, in on_packet_received
    result = self.prn(pkt)
  File "/root/IMSI-catcher/simple_IMSI-catcher.py", line 561, in find_imsi_from_pkt
    find_imsi(udpdata)
  File "/root/IMSI-catcher/simple_IMSI-catcher.py", line 490, in find_imsi
    t.register_imsi(gsm.arfcn, imsi1, imsi2, tmsi1, tmsi2, p)
  File "/root/IMSI-catcher/simple_IMSI-catcher.py", line 276, in register_imsi
    self.pfields(str(n), tmsi1, tmsi2, imsi1, str(self.mcc), str(self.mnc), str(self.lac), str(self.cell), p)
  File "/root/IMSI-catcher/simple_IMSI-catcher.py", line 193, in pfields
    self.output_function(cpt, tmsi1, tmsi2, imsi, imsicountry, imsibrand, imsioperator, mcc, mnc, lac, cell, now, packet)
  File "/root/IMSI-catcher/simple_IMSI-catcher.py", line 182, in output
    print("{:7s} ; {:10s} ; {:10s} ; {:17s} ; {:12s} ; {:10s} ; {:21s} ; {:4s} ; {:5s} ; {:6s} ; {:6s} ; {:s}".format(str(cpt), tmsi1, tmsi2, imsi, imsicountry.encode('utf-8'), imsibrand.encode('utf-8'), imsioperator.encode('utf-8'), str(mcc), str(mnc), str(lac), str(cell), now.isoformat()))
TypeError: unsupported format string passed to bytes.__format__

lapolis, 2021-07-13:
Step to make it work:
Step 1 - Install scapy for Python 2.7
Step 2 - Use Python 2.7 to run the code

Bonus point - additional step:
Step 3 - Wait few hours until I patch it for Python 3.9

Hope it is more clear now.

NuranFerzeli, 2021-07-13:
okey thanks for attention

lapolis, 2021-07-13:
This fix should work --> #131

Oros42, 2021-07-13:
Merge done

NuranFerzeli, 2021-07-13:
it works perfectly thanks a lot this is amazing project!. thanks for support

lapolis, 2021-07-13:
> it works perfectly thanks a lot this is amazing project!. thanks for support

Glad it does.

And sorry @Oros42 for the mess

closed 2021-07-13 by Oros42

alphafox02, 2021-07-13:
Just fyi - I dropped this as is into "DragonOS" which has python2/3 and had no issues running sudo python3 simple_IMSI-catcher.py --sniff - but I also had no issues before these updates. Either way, it works.

lapolis, 2021-07-13:
> Just fyi - I dropped this as is into "DragonOS" which has python2/3 and had no issues running sudo python3 simple_IMSI-catcher.py --sniff - but I also had no issues before these updates. Either way, it works. 

I was on DragonOS too. The latest one. The whole script was working well but it was crashing whenever there were some unicode in the country|brand|operator column.

I realised that when I used a Polish sim card from an operator that has the \u00f3 letter in the name. Not sure if I was in Python3 or 2 tho.