Oros42/IMSI-catcher #137: Not catching my own IMSI

Uriaz2002, 2021-09-23:
Hi mates,
I am trying to catch IMSI numbers on a specific frequency and it catches a lot of IMSI numbers but it does not catch my own IMSI (I have 4 mobile phones but it catches not even one of those). 
I have tried sniffing my own IMSI by using the syntax "python3 simple_IMSI-catcher.py --sniff --imsi="XXX XX XXXXXXXXXX" but it still does not sniff it.
Kindly help me if someone has a solution to it.
Using:
Ubuntu 20.04
python 3.8
USRP / Hackrf

r1bnc, 2021-09-23:
We have the same problem, I've even set my phone to use 2G only

np1621, 2021-10-13:
I think  that TSMI is used to promote confidentiality and protect IMSI numbers,  we must have use -a on  "python3 simple_IMSI-catcher.py" to see them. So I think its possible that our own IMSI could not came because of this, as well as a lot of them. Try script with -a and you will see many TSMI.

Sammca, 2021-11-19:
Whilst experiencing the same issue I've been trying to understand and my best guess is to do with tmsi caching. 

> According to 3GPP 33.102 v9.30, TMSI should be changed periodically. But it is rarely done.

It appears as though tmsi are not as temporary as intended and if your phone has already attached to the tower/network prior to beginning the program, your imsi may never show up. This is set by the network so may vary and may also explain why I see such a large number of foreign/roaming devices. I have yet to find a way to flush/reset an imsi on a network.

However I have set up a openBTS station and have been unable to catch devices registering to this network, even when setting the tmsi to expire after 5 minutes. I'm still testing this, and intend to test the TMSI theory with a device I can obtain the current tmsi from and use the -a argument posted by np1621 above. Any feedback from @Oros42 would be much appreciated.

ChillVibesMushroom, 2022-06-25:
> We have the same problem, I've even set my phone to use 2G only

What GSM band is 2G?

ChillVibesMushroom, 2022-06-25:
We just lower our band then?

PARAMJAI, 2023-02-17:
can anyone share the code, I am having errors with the code provided by Oros42.