Oros42/IMSI-catcher #141: Unable to find any supported SDR Device

Warhammar, 2021-11-08:
INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_3.15.0.0-5
[ERROR] avahi_service_browser_new() failed: Bad state
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 385, in <module>
    main()
  File "/usr/bin/grgsm_livemon", line 361, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/bin/grgsm_livemon", line 113, in __init__
    args="numchan=" + str(1) + " " + str(grgsm.device.get_default_args(args))
  File "/usr/lib/python3/dist-packages/grgsm/device.py", line 55, in get_default_args
    raise RuntimeError("Unable to find any supported SDR devices")
RuntimeError: Unable to find any supported SDR devices

I have Realtek RTL2832U DVR Dongle. 
What am i doing wrong_

PARAMJAI, 2023-02-17:
After giving a good try I am wondering now if this IMSI Catcher had ever worked for anyone?

pablo-01, 2025-12-03:
> After giving a good try I am wondering now if this IMSI Catcher had ever worked for anyone?

I've just tried and it does seem to work. 
I have followed the installation instructions including making sure I use the recommended OS and python version etc. 

For the 'Unable to find any supported SDR' devices - you first need make sure correct drivers are installed or on Ubuntu blacklist the existing driver before using your device with this software.   

>'blacklisting default kernel modules on an Ubuntu system so that your Software Defined Radio (SDR) device can be accessed by SDR applications instead of the default digital TV tuner drivers'  

or if you done this already try using different usb port for your SDR.

flanneryb, 2025-12-03:
Hello,

It does work, you need to install the correct dependencies, you also need
to use it on a network that will allow it.